SmartThingsCommunity/st-device-sdk-c-ref #25: Error while esp8266 build

anandkamat05, 2020-06-30:
Getting this error when attempting to build st_switch for esp32.
Provided the console log showing the error.
I have only simply followed the documentation provided in the repository (https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/tree/master/apps/esp32/st_switch). Can someone please help me with this?

command used: ./build.sh esp32 st_switch

Generating libspi_flash.a.sections_info
Generating libspiffs.a.sections_info
Generating libsrc.a.sections_info
Generating libtcp_transport.a.sections_info
Generating libtcpip_adapter.a.sections_info
Generating libulp.a.sections_info
Generating libunity.a.sections_info
Generating libvfs.a.sections_info
Generating libwear_levelling.a.sections_info
Generating libwifi_provisioning.a.sections_info
Generating libwpa_supplicant.a.sections_info
Generating libxtensa-debug-module.a.sections_info
Generating esp32.project.ld
LD build/smart_switch_demo.elf
/home/anand/st-device-sdk-c-ref/apps/esp32/st_switch/build/src/libsrc.a(iot_log_file.o):(.literal._iot_log_file_load_header+0x14): undefined reference to `iot_log_read_flash'
/home/anand/st-device-sdk-c-ref/apps/esp32/st_switch/build/src/libsrc.a(iot_log_file.o): In function `_iot_log_file_load_header':
/home/anand/st-device-sdk-c-ref/iot-core/src/iot_log_file.c:94: undefined reference to `iot_log_read_flash'
/home/anand/st-device-sdk-c-ref/apps/esp32/st_switch/build/src/libsrc.a(iot_log_file.o): In function `_iot_log_read_bytes':
/home/anand/st-device-sdk-c-ref/iot-core/src/iot_log_file.c:94: undefined reference to `iot_log_read_flash'
collect2: error: ld returned 1 exit status
make: *** [/home/anand/st-device-sdk-c-ref/bsp/esp32/make/project.mk:483: /home/anand/st-device-sdk-c-ref/apps/esp32/st_switch/build/smart_switch_demo.elf] Error 1

stdk-scm, 2020-06-30:
Dear anandkamat05

Thanks for your reporting
We will check & fix it as soon as possible

stdk-scm, 2020-06-30:
Dear anandkamat05

Please use following changes to fix this issue
The official patch-set will be apply on the next release

diff --git a/apps/esp32/st_switch/sdkconfig b/apps/esp32/st_switch/sdkconfig
index 4389a83..8ad18ab 100644
--- a/apps/esp32/st_switch/sdkconfig
+++ b/apps/esp32/st_switch/sdkconfig
@@ -694,10 +694,7 @@ CONFIG_STDK_IOT_CORE_LOG_LEVEL_WARN=y
 CONFIG_STDK_IOT_CORE_LOG_LEVEL_INFO=y
 CONFIG_STDK_IOT_CORE_LOG_LEVEL_DEBUG=
 CONFIG_STDK_IOT_CORE_SUPPORT_STNV_PARTITION=
-CONFIG_STDK_IOT_CORE_LOG_FILE=y
-CONFIG_STDK_IOT_CORE_LOG_FILE_RAM_ONLY=y
-CONFIG_STDK_IOT_CORE_LOG_FILE_FLASH_WITH_RAM=
-CONFIG_STDK_IOT_CORE_LOG_FILE_RAM_BUF_SIZE=1024
+CONFIG_STDK_IOT_CORE_LOG_FILE=
 CONFIG_STDK_IOT_CORE_BSP_SUPPORT_ESP8266=
 CONFIG_STDK_IOT_CORE_BSP_SUPPORT_ESP32=y
 CONFIG_STDK_IOT_CORE_OS_SUPPORT_FREERTOS=y

Thanks

anandkamat05, 2020-07-01:
I changed the file sdkconfig ( at st-device-sdk-c-ref/apps/esp32/st_switch/sdkconfig) as suggested. 
Removed the following lines:
-CONFIG_STDK_IOT_CORE_LOG_FILE=y
-CONFIG_STDK_IOT_CORE_LOG_FILE_RAM_ONLY=y
-CONFIG_STDK_IOT_CORE_LOG_FILE_FLASH_WITH_RAM=
-CONFIG_STDK_IOT_CORE_LOG_FILE_RAM_BUF_SIZE=1024

and added this as in the comment.
+CONFIG_STDK_IOT_CORE_LOG_FILE=

Still getting the same error.

closed 2020-07-02 by anandkamat05