SmartThingsCommunity/st-device-sdk-c-ref #27: Onboarding Error ESP32 Switch

anandkamat05, 2020-07-02:
I am attempting to run the st_switch program from this repo on an ESP32 chip. 
After flashing the code onto the chip, i wasn't able to detect the device using the smart things app. I then realized the device's wifi ssid wasn't visible. I have made no changes to the code yet.
I used the monitor to detect probable issues and got this after flashing:
(command used "./build.sh esp32 st_switch flash monitor")

MONITOR

I (309) cpu_start: Pro cpu up.
I (309) cpu_start: Application information:
I (309) cpu_start: Project name:     smart_switch_demo
I (312) cpu_start: App version:      v1.0.0-191-gd5f6ec6-dirty
I (319) cpu_start: Compile time:     Jul  2 2020 21:16:21
I (325) cpu_start: ELF file SHA256:  65496e38ea8136e7...
I (331) cpu_start: ESP-IDF:          v3.3-rc-10-g8f5cf4754
I (337) cpu_start: Single core mode
I (341) heap_init: Initializing. RAM available for dynamic allocation:
I (348) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (354) heap_init: At 3FFB87F8 len 00027808 (158 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (367) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (373) heap_init: At 4008EFB4 len 0001104C (68 KiB): IRAM
I (379) cpu_start: Pro cpu start user code
I (61) cpu_start: Starting scheduler on PRO CPU.
**E (65) [IoT]: iot_api_onboarding_config_load(498) > [mnId] wrong onboarding config value detected: MNID
E (65) [IoT]: st_conn_init(1092) > failed loading onboarding profile (-5)
fail to create the iot_context**
I (75) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (85) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (95) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (105) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (115) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (125) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3

Can i do something to fix this? I am closely followed the documentation provided for st_switch, am using esp-idf v3.3 and the toolchain as recommended in the readme files.

Kwang-Hui, 2020-07-02:
Hi, you need to download your project's onboarding_config.json and put it into your project source code.
Please refer https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/tree/master/apps/esp32/st_switch `3.6. Download onboarding_config.json`

anandkamat05, 2020-07-03:
Sorry, I missed that.

closed 2020-07-03 by anandkamat05