SmartThingsCommunity/st-device-sdk-c-ref #29: Unable to register device to samsung acount

anandkamat05, 2020-07-03:
Error while registering **st_switch** using ESP32 to Samsung account. Code used is exactly the same as in this repository (along with this issue correction https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/issues/25). Implementation details provided by smart things used. (esp-idf version 3.3 and toolchain provided in this repository used for the project). Device was successfully discovered in the smart things app but was unable to get registered to my Samsung account. 
While using a monitor, the following output was printed after selecting the wifi network the device will connect to (error in bold).

I (67111) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [5]
I (67111) [IoT]: _es_wifi_prov_parse(1327) > no macAddress
I (67121) [IoT]: _es_wifi_prov_parse(1350) > ssid: test_wifi
I (67121) [IoT]: _es_wifi_prov_parse(1352) > mac addr: 00:00:00:00:00:00
I (67131) [IoT]: _es_json_parse_string(145) > failed to find 'locationId'
I (67131) [IoT]: _es_cloud_prov_parse(1425) > no locationId
I (67141) [IoT]: _es_json_parse_string(145) > failed to find 'roomId'
I (67151) [IoT]: _es_cloud_prov_parse(1446) > no roomId
I (67161) [IoT]: _es_cloud_prov_parse(1466) > brokerUrl: mqtt-apnortheast2.api.smartthings.com:8883
I (67161) [IoT]: _es_cloud_prov_parse(1467) > deviceName : SmartSwitch
I (67171) [IoT]: _es_wifiprovisioninginfo_handler(1594) > provisioning success
I (67181) [IoT]: http_msg_handler(338) > post cmd[5] ok
I (67201) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [6]
I (67201) [IoT]: http_msg_handler(338) > post cmd[6] ok
I (67211) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:3/3
I (67371) wifi: station: 26:9f:ca:0e:6f:42 leave, AID = 1, bss_flags is 134243, bss:0x3ffc9bc8
I (67371) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (67371) [IoT]: event_handler(139) > station:26:9f:ca:0e:6f:42leave, AID=1
I (68211) [IoT]: _do_update_timeout(161) > Current timeout : 100000 for 4
I (68211) [IoT]: _do_status_report(265) > Call usr status_cb with 2/2
status: 2, stat: 2
I (68211) [IoT]: _do_iot_main_command(332) > curr_main_cmd:1, curr_main_state:3/4
I (68221) [IoT]: iot_bsp_wifi_set_mode(212) > iot_bsp_wifi_set_mode = 2
I (68231) [IoT]: iot_bsp_wifi_set_mode(264) > [esp32] current mode=3 need to call esp_wifi_stop
I (68241) [IoT]: event_handler(98) > SYSTEM_EVENT_STA_STOP
**E (68241) event: system_event_ap_stop_handle_default 223 esp_wifi_internal_reg_rxcb ret=0x3014**
I (68251) [IoT]: event_handler(126) > SYSTEM_EVENT_AP_STOP
I (68261) wifi: flush txq
I (68261) wifi: stop sw txq
I (68261) wifi: lmac stop hw txq
I (68271) [IoT]: iot_bsp_wifi_set_mode(271) > AP Mode stopped
I (68771) wifi: mode : sta (3c:71:bf:af:a8:00)
I (68771) [IoT]: iot_bsp_wifi_set_mode(305) > connect to ap SSID:test_wifi
I (69131) wifi: new:<4,1>, old:<1,0>, ap:<255,255>, sta:<4,1>, prof:1
I (70111) wifi: state: init -> auth (b0)
I (70121) wifi: state: auth -> assoc (0)
I (70131) wifi: state: assoc -> run (10)
I (70241) wifi: connected with test_wifi, channel 4, 40U, bssid = c*:3*:e*:7*:*6:*c
I (70241) wifi: pm start, type: 1

I (70241) [IoT]: event_handler(145) > event_handler = 4
I (72351) event: sta ip: 192.***.***.**, mask: 255.***.***.*, gw: 192.***.*.*
I (72351) [IoT]: event_handler(113) > got ip:192.xxx.x.xxx rssi:-*****
I (72351) [IoT]: iot_bsp_wifi_set_mode(310) > AP Connected
I (72361) [IoT]: _clear_listen_socket(44) > Shutting down listen socket
I (72361) [IoT]: es_http_deinit(229) > http tcp deinit complete!
I (72371) [IoT]: iot_easysetup_deinit(462) > IOT_STATE_PROV_ES_DONE
I (72381) [IoT]: _do_iot_main_command(332) > curr_main_cmd:3, curr_main_state:3/4
I (72391) [IoT]: _do_cmd_tout_check(813) > New state updated for 4
I (72391) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:4/4
I (72401) [IoT]: _do_state_updating(1483) > the state changes to IOT_STATE_CLOUD_REGISTERING
I (72411) [IoT]: _do_update_timeout(161) > Current timeout : 900000 for 6
I (72421) [IoT]: _do_status_report(265) > Call usr status_cb with 8/1
status: 8, stat: 1
I (72431) [IoT]: _do_iot_main_command(332) > curr_main_cmd:4, curr_main_state:4/6
I (72471) [IoT]: iot_es_connect(792) > connect_type: registration
I (72471) [IoT]: _iot_es_mqtt_connect(631) > url: mqtt-apnortheast2.api.smartthings.com, port: 8883
I (72471) [IoT]: _iot_es_mqtt_connect(640) > mqtt connect,
id : *********************************
username : ST******
password : ** password printed here **
I (72511) [IoT]: _iot_net_tls_connect(160) > Loading the CA root certificate *********
I (74961) [IoT]: iot_es_connect(805) > MQTT connect success
I (75511) [IoT]: _do_cmd_tout_check(813) > New state updated for 6

stdk-scm, 2020-07-03:
Dear  anandkamat05 

Thanks for your interest about our STDK

1. Could you please wait for more time about 1 ~ 2 min. (Sometimes Server replied it late)
2. Could you please share to us about full logs 
   * eamil : stdk@samsung.com
   * we need to get followings
     - id : xxxxxxxxxxxxx
     - username : ST--------

anandkamat05, 2020-07-05:
Waiting doesn't work.

If you are unable to reproduce the error, it might be due to incompatible versions of esp-idf and toolchain. Can you please verify them?

Have sent complete logs to the email address provided in the comment above.

stdk-scm, 2020-07-06:
Dear anandkamat05 

We use following toolchain from ESP

xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

And please refer to following our today working logs

iot_status: 2, lv: 4
I (31258) [IoT]: http_msg_handler(356) > get cmd[0] ok
I (31338) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [1]
I (31608) [IoT]: _es_keyinfo_handler(570) > master secret generation success
I (31608) [IoT]: http_msg_handler(338) > post cmd[1] ok
I (31758) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [2]
I (31758) [IoT]: _es_confirminfo_handler(888) > otmSupportFeature = 0
I (31768) [IoT]: _es_confirm_check_manager(753) > IOT_STATE_PROV_CONFIRMING
I (31768) [IoT]: iot_state_update(196) > Trigger PROV_CONFIRM
I (31778) [IoT]: _es_confirm_check_manager(767) > There is no confirmation request. The check is skipped
I (31788) [IoT]: http_msg_handler(338) > post cmd[2] ok
I (31798) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:1/1
I (31808) [IoT]: _do_state_updating(1458) > the state changes to IOT_STATE_PROV_CONFIRM
I (31808) [IoT]: _do_update_timeout(161) > Current timeout : 100000 for 3
I (31818) [IoT]: _do_cmd_tout_check(813) > New state updated for 3
I (31828) [IoT]: st_conn_start(1637) > st_conn_start done (0)
I (31838) [IoT]: _iot_easysetup_gen_get_payload(140) > waiting.. response for [4]
I (31838) [IoT]: iot_bsp_wifi_set_mode(212) > iot_bsp_wifi_set_mode = 1
I (34298) [IoT]: event_handler(145) > event_handler = 1
I (34358) [IoT]: http_msg_handler(356) > get cmd[4] ok
I (40548) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [5]
I (40548) [IoT]: _es_wifi_prov_parse(1327) > no macAddress
I (40558) [IoT]: _es_wifi_prov_parse(1350) > ssid: XXXXXXXX
I (40558) [IoT]: _es_wifi_prov_parse(1352) > mac addr: 00:00:00:00:00:00
I (40568) [IoT]: _es_json_parse_string(145) > failed to find 'locationId'
I (40578) [IoT]: _es_cloud_prov_parse(1425) > no locationId
I (40578) [IoT]: _es_json_parse_string(145) > failed to find 'roomId'
I (40588) [IoT]: _es_cloud_prov_parse(1446) > no roomId
I (40598) [IoT]: _es_cloud_prov_parse(1466) > brokerUrl: mqtt-useast1.api.smartthings.com:8883
I (40598) [IoT]: _es_cloud_prov_parse(1467) > deviceName : SmartSwitch_0
I (40618) [IoT]: _es_wifiprovisioninginfo_handler(1594) > provisioning success
I (40618) [IoT]: http_msg_handler(338) > post cmd[5] ok
I (40638) [IoT]: _iot_easysetup_gen_post_payload(247) > waiting.. response for [6]
I (40638) [IoT]: http_msg_handler(338) > post cmd[6] ok
I (40648) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:3/3
I (40708) wifi: station: 8c:f5:a3:74:70:f1 leave, AID = 1, bss_flags is 134243, bss:0x3ffc8298
I (40708) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (40708) [IoT]: event_handler(139) > station:8c:f5:a3:74:70:f1leave, AID=1
I (41648) [IoT]: _do_update_timeout(161) > Current timeout : 100000 for 4
I (41648) [IoT]: _do_status_report(265) > Call usr status_cb with 2/2
iot_status: 2, lv: 2
I (41648) [IoT]: _do_iot_main_command(332) > curr_main_cmd:1, curr_main_state:3/4
I (41658) [IoT]: iot_bsp_wifi_set_mode(212) > iot_bsp_wifi_set_mode = 2
I (41668) [IoT]: iot_bsp_wifi_set_mode(264) > [esp32] current mode=3 need to call esp_wifi_stop
I (41678) [IoT]: event_handler(98) > SYSTEM_EVENT_STA_STOP
E (41678) event: system_event_ap_stop_handle_default 223 esp_wifi_internal_reg_rxcb ret=0x3014
I (41688) [IoT]: event_handler(126) > SYSTEM_EVENT_AP_STOP
I (41698) wifi: flush txq
I (41698) wifi: stop sw txq
I (41698) wifi: lmac stop hw txq
I (41708) [IoT]: iot_bsp_wifi_set_mode(271) > AP Mode stopped
I (42208) wifi: mode : sta (84:0d:8e:e6:80:28)
I (42208) [IoT]: iot_bsp_wifi_set_mode(305) > connect to ap SSID:syshot
I (43538) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (44518) wifi: state: init -> auth (b0)
I (44558) wifi: state: auth -> assoc (0)
[... 31 lines cut ...]
I (52638) [IoT]: _iot_net_show_status(81) > [1593999575] network socket status: readable 0 writable 0 sock_err 0 errno 128
I (52648) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:6/6
I (52658) [IoT]: _do_update_timeout(161) > Current timeout : 100000 for 7
I (52668) [IoT]: _do_status_report(265) > Call usr status_cb with 8/6
iot_status: 8, lv: 6
I (52678) [IoT]: _do_iot_main_command(332) > curr_main_cmd:6, curr_main_state:6/7
I (52718) [IoT]: iot_es_connect(740) > connect_type: log-in
I (52718) [IoT]: _iot_es_mqtt_connect(631) > url: mqtt-useast1.api.smartthings.com, port: 8883
I (52718) [IoT]: _iot_es_mqtt_connect(640) > mqtt connect,
id : c4b8de72-a013-41a0-88ff-XXXXXXXX
username : 71473817-ba1e-4ab5-9aac-XXXXXXXX
password : eyJhbGciOiJFZERTQSIsImt0eSI6Ik9LUCIsImNXXXXXXXX
I (52758) [IoT]: _iot_net_tls_connect(160) > Loading the CA root certificate 1339@0x3ffc6d3c
I (54988) [IoT]: iot_es_connect(758) > MQTT connect success
I (55818) [IoT]: _do_cmd_tout_check(813) > New state updated for 7
I (55818) [IoT]: _do_iot_main_command(332) > curr_main_cmd:9, curr_main_state:7/7
I (55818) [IoT]: _do_update_timeout(161) > Current timeout : 100000 for 8
[... cut ...]
I (55838) [IoT]: _do_iot_main_command(332) > curr_main_cmd:0, curr_main_state:7/8
I (55848) [IoT]: iot_cap_call_init_cb(1070) > Call init_cb for switch capability
Sequence number return : 1
I (55858) [IoT]: _do_cmd_tout_check(813) > New state updated for 8
I (55858) [IoT]: _publish_event(858) > publish event, topic : /v1/deviceEvents/71473817-ba1e-4ab5-9aac-XXXXXXXX, payload :
{"deviceEvents":[{"component":"main","capability":"switch","attribute":"switch","value":"on","providerData":{"sequenceNumber":1,"timestamp":"1593999578242"}}]}

Kwang-Hui, 2020-07-06:
@anandkamat05 Could you let me know SmartThings mobile application version?
_SmartThings app -> Settings -> About SmartThings_

anandkamat05, 2020-07-06:
The latest version of SmartThings is already installed. I use a Galaxy S10.
SmartThings Version 1.7.50-21.
I also have a simple st_switch working fine on a esp8266 in the same app (different developer project). 

Since your logs also show the error "**event: system_event_ap_stop_handle_default 223 esp_wifi_internal_reg_rxcb ret=0x3014**" and still manages to work fine, i doubt that is the cause.

The last line **[IoT]: _do_cmd_tout_check(813) > New state updated for 6** is where its getting stuck. 
May i please know which version of esp-idf you are using? The toolchain version i have matches yours.

This may be completely unrelated but would still like to throw it in for consideration. The project was created by a different admin and i was added as a developer (organization's project). At first i was unable to setup the device from my samsung account linked to that project as the setup apps in my phone's app didn't show our device (we had to add/configure the device on SmartThings App using the admin's account). Once i created a private project (separate from the organization's project) was i able to see the organization's setup apps and devices in my SmartThings app (along with the private project's device). Since the issue relates to registering the device to the samsung account thought i should mention this. Even though i was added to the organization's project as a developer, i wasn't able to setup the device from my account until i created a private project, which showed me organization's devices too.

Kwang-Hui, 2020-07-06:
> The latest version of SmartThings is already installed. I use a Galaxy S10.
> SmartThings Version 1.7.50-21.
> I also have a simple st_switch working fine on a esp8266 in the same app (different [...]

I think **"event: system_event_ap_stop_handle_default 223 esp_wifi_internal_reg_rxcb ret=0x3014"** is not a root cause. I could see the same log print. but works fine.

I'm using [this](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz) version which is mentioned at README.md. But I believe that doesn't causes this problem.

As I know, organization developer can't join admin's project.
So I recommend to use organization admin's account for testing or private project for now.

anandkamat05, 2020-07-06:
Alright so i replaced the device credentials (serial np, public key and private key) from an older working project (linking my device to that older project) and it is working fine as is. It is only when i add the newly generated keys to the device, do i experience this issue.
Could you try this on your end. Generate new keys using the keygen provided, add them as device identity and then adding this device to the app. Only when i am adding new keys as device identity do i face the issue.  

Also, the developer workspace doesn't let us delete or edit old keys added. There are only a finite number of keys which can be added but the older ones cannot be deleted. Last time this happened we had to delete our account and renew it to be able to add new devices. Currently we have carefully only added one device identity to each project we have. Also it would be helpful to be able to see which device credentials are currently used by the project.

Kwang-Hui, 2020-07-08:
> Alright so i replaced the device credentials (serial no., public key and private key) from an older working project (linking my device to that older project) and it is working fine now as is. It is only when i add the newly generated keys to the device, do i experience this issue.
> Could you try this on your end. Generate new keys using the keygen provided, add them as device identity on the developer workspace and then try adding this device to the app. Only when i am adding new keys as device identity do i face the issue.
> 
> Also, the developer workspace doesn't let us delete or edit old keys. There are only a finite number of keys (test devices) which can be added but the older ones cannot be deleted. Last time this happened we had to delete our account and renew it to be able to add new devices. Currently we have carefully only added one device identity to each project. Also it would be helpful to be able to see which device credentials are currently used by the project.

Actually I can't reproduce it on my side. Sorry about that.
As I know managing test devices at developer workspace would be part of commercial program feature and it is under test.
I think you can get it at August.
https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/wiki/Get-ready-to-commercialize-your-devices

Thank you.

anandkamat05, 2020-07-08:
I understand. Could you please try this last Hail Mary to reproduce this issue? 
Could you try creating a new project from the developer workspace and then just follow the steps to add a new device (with fresh keys) and see if the device can be set up on the SmartThings App. 
The issue we are facing is just confined to the projects (commercial as well as private) we created in the last 5-7 days. The older projects are still working fine. 
Sorry to keep troubling you about this but we are stuck due to this issue.
We appreciate all the effort you have put in this for us.

Kwang-Hui, 2020-07-27:
Hi @anandkamat05  I'm sorry for late response.
I tested on my side today with below steps.
it works fine for me.

1) create new project
2) add first device identity 
3) test device's onboarding and control (st_switch example)
4) add second device identity (new serial number and public key)
5) test device's onboarding and control (st_switch example)

Test environment is 
* Android ST app (1.7.50-21)
* SmartThings Device SDK ver 1.1.18
* Samsung Account Region: US (broker url: mqtt-useast1.api.smartthings.com)

I'll check if there is an issue on asia-pacific region server (mqtt-apnortheast2.api.smartthings.com)

Thanks.

stdk-scm, 2020-07-27:
Hi @anandkamat05 . we could find problem from your issue request.
there is a problem while onboarding to asia-pacific regional server.
we will fix this issue and ping to you once this has been fixed.

thanks for your report again.

Kwang-Hui, 2020-07-27:
Hi @anandkamat05 . we could find problem from your issue request.
 there is a problem while onboarding to asia-pacific regional server.
 we will fix this issue and ping to you once this has been fixed.

thanks for your report again.

anandkamat05, 2020-07-27:
Thank you so much. I appreciate it. We will be waiting for the update.

jsp0929, 2020-07-31:
I have the same problem. It always stops at 99% during the on-boarding process. Please notify here when you fix it.

jsp0929, 2020-07-31:
> Hi @anandkamat05 . we could find problem from your issue request.
> there is a problem while onboarding to asia-pacific regional server.
> we will fix this issue and ping to you once this has been fixed.
> 
> thanks for your report again.

I have the same problem. It always stops at 99% during the on-boarding process. Please notify here when you fix it.

Kwang-Hui, 2020-07-31:
> I have the same problem. It always stops at 99% during the on-boarding process. Please notify here when you fix it.

@jsp0929 
Thank for reporting issue.
Our team found that this problem requires mobile application update.
The next mobile application update is scheduled at mid of Aug. and it will have corresponding  changes to Developer workspace and SmartThings Device SDK commercial program features.

I'm sorry about your inconvenience and I will let you know about the fix is available.
meanwhile you can test your device by creating US region samsung account.

anandkamat05, 2020-08-04:
Just an update to let you know that the same issue is observed when you use an US account as well.
Hope you fix this soon. Eagerly awaiting the update.

Kwang-Hui, 2020-08-05:
Hello @anandkamat05 @jsp0929 
This issue will be fixed with updated version of SmartThings mobile application.
That would be deployed starting from end of this week and could takes some time to all of users depends on market channels.
The version number would be "v1.7.51" for Android, "v1.6.51" for iOS

your bug report was really helpful for us.
much appreciated.

Kwang-Hui, 2020-08-12:
There might be little delay on App update deployment. I think it doesn't take a long time.
sorry about that.

f34rdotcom, 2020-08-14:
Any news? I am still stalled on development on my project till the app is updated :(

vnk63, 2020-08-16:
> There might be little delay on App update deployment. I think it doesn't take a long time.
> sorry about that.
We have been eagerly waiting for six weeks now. We have noticed the updated code on github and it would be exciting to test it. Is the updated version of SmartThings mobile application  v1.7.51, still scheduled for deployment in August considering the delays you have mentioned; or will it take longer? A **_rough time estimate_** for the successful onboarding to asia-pacific regional server with mobile app v1.7.51 would help.

Kwang-Hui, 2020-08-17:
Hi All,
SmartThings mobile application update has re-scheduled at end of this week.
I'm sorry for your inconvenience.

vnk63, 2020-08-23:
Tested successfully in India after the SmartThings mobile Application update (Version 1.7.51.42). It appears that the on-boarding problem while connecting to Asia-Pacific Regional Server has been resolved. Thank you, @Kwang-Hui et al.

Kwang-Hui, 2020-08-24:
Hi All,

As @vnk63 mentioned, SmartThings mobile application has been released. Please use it for your test devices.

f34rdotcom, 2020-08-24:
The app is out! I updated and got back to testing. I am still having problems registering my device. I am using an ESP32 dev kit and have based my example using the switch_example. I have tried deleting all of my "Workspace" details and rebuilding as well as other tests but I keep having the same error. I even rebuilt my pub/private key.

On thing I noticed in my onboarding_config.json is that the id as a trailling "_" so when my device shows up as a wifi node it has two "__" and shows as "AlarmDecoder__E40A0f1006g...." 

[image]

I tried tuning on debugging but as best I can tell it never actually connects to the WiFi on the ESP32 to provision so it seems to be failing in finding it as a WiFi node to connect to?

Thanks in advance.
 SM.

Kwang-Hui, 2020-08-24:
> The app is out! I updated and got back to testing. I am still having problems registering my device. I am using an ESP32 dev kit and have based my example using the switch_example. I have tried [...]

I think the two "_" is not a problem.
Could you confirm you have registered serial number and public key of your new test device?
You could see your test device list from `Developer Workspace` -> `Project` -> `Test devices`

f34rdotcom, 2020-08-25:
Yes :( several times. I tried making new keys I did everything I can think of. I got rid of the '__' by changing the device name(s) to try and simplify my tests and nothing so far has worked :(

I connected my phone to my system to debug. Also on my ESP32 device I never see any connection from my phone.

`
$> adb logcat | grep oneconnect
"CloudLogger.send - log ="{
   "apconnected":{
      "connectedfreq":2437,
      "connectedssid":"($)-T(10)"
   },
   "apdual":{
      "isconnecteddual":false,
      "issupport5g":false
   },
   "aphomeap":{
      "auth":"",
      "capabilities":"",
      "enc":"",
      "freq":0,
      "hidden":"",
      "pwexists":"",
      "ssid":"UNKNOWN"
   },
   "aprssi":{
      "devicehomerssi":"0",
      "mobilehomerssi":"-57",
      "softaprssi":-20
   },
   "apwifiselect":{
      "devcnt":0,
      "devtimeout":false,
      "scancnt":0,
      "state":"AUTO"
   },
   "bintype":"user",
   "bondingresult":"",
   "btfwver":"",
   "connectedothercount":0,
   "ctry":"US",
   "currsession":"",
   "currtime":"2020-08-24 20:08:41",
   "d2dping":-1,
   "dnstime":206,
   "dumpfileid":"",
   "dumpfilepath":"",
   "elapdiscovery":0,
   "elaptime":108974,
   "entry":"RETRY",
   "errcode":"81-001-MBL",
   "errcodeMain":"81",
   "errcodeSub":"001",
   "esconntype":"WIFI",
   "eslog":"08-24 20:08:34.004 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.DiscoveryHandler - [D2S] found target:[Name]($)-A(28)[Pid]null[Pool]false[DeviceType]SAMSUNG_OCF_SETUP[Discover]_WIFI[Services]None[Connected]false[ConType]NONE[size()]1[OCFDiscovery]_WIFI[OCFOwned]-1\n08-24 20:08:34.052 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.DiscoveryHandler - [D2S] found target:[ESDT]SamsungStandard_E4[Name]($)-A(28)[DiscoveryType]_WIFI[Protocol]OCF[ManualSetup]false[ShpSetupFromDeviceCard]false[StandardSsidInfo][Name]($)-A(9)[Ver]4[MnId]0AOf[SetuId]100[ExtType]6[ExtValue]vheAM1NK[ApConnected]false[WiredConnected]false\n08-24 20:08:36.149 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.startDiscovery - scantype:_WIFI_FULL_MANUAL_SETUP, flush:false\n08-24 20:08:36.455 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.DiscoveryHandler - [D2S] found target:[Name]($)-A(28)[Pid]null[Pool]false[DeviceType]SAMSUNG_OCF_SETUP[Discover]_WIFI[Services]None[Connected]false[ConType]NONE[size()]1[OCFDiscovery]_WIFI[OCFOwned]-1\n08-24 20:08:36.497 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.DiscoveryHandler - [D2S] found target:[ESDT]SamsungStandard_E4[Name]($)-A(28)[DiscoveryType]_WIFI[Protocol]OCF[ManualSetup]false[ShpSetupFromDeviceCard]false[StandardSsidInfo][Name]($)-A(9)[Ver]4[MnId]0AOf[SetuId]100[ExtType]6[ExtValue]vheAM1NK[ApConnected]false[WiredConnected]false\n08-24 20:08:37.417 - [-8]oneconnect[1.7.51.42_01][CORE].EasySetupDiscoveryManager.stopDiscovery - type:[SamsungStandard_E4], timeout:300000,includeCloud:false, excludeL3:true\n08-24 20:08:37.880 - [-8]oneconnect[1.7.51.42_01][CORE].[EasySetup]EasySetupMcSignInState.EasySetupMcSignIn - IN\n08-24 20:08:37.882 - [-8]oneconnect[1.7.51.42_01][CORE].[EasySetup]GetAuthState.GetAuthState - IN\n08-24 20:08:39.485 - [-8]oneconnect[1.7.51.42_01][CORE].[EasySetup]OCFEasySetupProtocol.QcEasySetupHandler - GET_ACCESSTOKEN\n08-24 20:08:40.708 - [-8]oneconnect[1.7.51.42_01][CORE].[EasySetup]GetCapiEncryptedKeyState.GetCapiEncryptedKeyState - IN\n",
   "esver":"UNKNOWN",
   "gattstate":{
      "blereport":"",
      "connstate":"NONE"
   },
   "helpcardopencount":0,
   "histories":[
      {
         "step":"INTRO",
         "time":105605
      },
      {
         "step":"SAMSUNGACCOUNT",
         "time":3361
      }
   ],
   "installer":"com.sec.android.app.samsungapps",
   "isd2ddevice":"false",
   "isemailid":"true",
   "isfileupload":false,
   "isrssifeatureon":"true",
   "isuserinputpin":"false",
   "lang":"en",
   "manualretrycount":1,
   "manufacturedyear":0,
   "mfr":"samsung",   
   "multidevicedetectedcount":-1,
   "os":"android",
   "otmreason":"UNKNOWN",
   "otmsupportfeature":0,
   "ownmatch":"true",
   "prevsession":"",
   "prevtgtstatus":"",
   "provisioningresults":[
      
   ],
   "registeringstate":"",
   "rescan":"false",
   "result":"FAIL",
   "retrycnt":0,
   "scanresultcount":0,
   "sn":{
      "number":"",
      "type":"SCANNING"
   },
   "stver":"1.7.51.42",
   "tgtcat":"SamsungStandardSsid",
   "tgtfwver":"UNKNOWN",
   "tgtlogid":"",
   "tgtprot":"OCF",
   "tgtssid":"AD2IOTV10_E40AOf1006vheAM1NK",
   "tgttype":"_E4",
   "tokenretrycount":0,
   "ultrasound":{
      "permission":"",
      "result":""
   },
   "wifihubcombi":"UNKNOWN"
}`

Kwang-Hui, 2020-08-25:
> Yes :( several times. I tried making new keys I did everything I can think of. I got rid of the '__' by changing the device name(s) to try and simplify my tests and nothing so far has worked :(
>
> I [...]

Could you let me know your SmartThings mobile application version string? it should be `1.7.51.42` if you are using android

kolos, 2020-08-25:
Hi,

I have the same "No device matches the serial number" (Error code: 81-001) issue.

I dont have organization, testing in the "Private workspace".
[image]

Device added:
[image]

SmartThings app version: 1.7.51.42

Thanks,
  Kolos

grobebar, 2020-08-25:
I'm having same issue with ESP8266 device and switch example.
When I looked at my adb logs I found things like this with one line standing out:

**08-25 10:40:46.814 15825 16181 W [-8]oneconnect[1.7.51.42_01][CORE]: SmartThingsEnergyController.updatedDevicePlatformInfo - not supported st energy service (blacklisted) deviceId : ($)2cecb532**

and since it shows following I have a suspicion it might b this device but what should I look for to confirm if this is indeed the ESP8266 Switch example I'm trying to connect and not some neighbor's device ;)
08-25 10:40:46.814 15825 13016 I [-8]oneconnect[1.7.51.42_01][CORE]: DeviceCloud.getRemoteRepresentation - uri : **/capability/switch/main/0**

```
08-25 10:40:46.811 15825 15825 I [-8]oneconnect[1.7.51.42_01][CORE]: BoardModeManager.deviceStateUpdated -  - ($)
08-25 10:40:46.812 16706 15020 I [-8]oneconnect[1.7.51.42_01][MAIN_UI]: Dash@DataHandler.handleDeviceItemDataMessage - Event: UPDATE
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: InternalSettingsManager.getFromSettingDB -  - ($)
08-25 10:40:46.814 15825 16181 W [-8]oneconnect[1.7.51.42_01][CORE]: SmartThingsEnergyController.updatedDevicePlatformInfo - not supported st energy service (blacklisted) deviceId : ($)2cecb532
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: HomeCareWizardController.updatedDevicePlatformInfo - not supported hcw deviceId : ($)2cecb532
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: InternalSettingsManager.getFromSettingDB -  - ($)
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: InternalSettingsManager.getFromSettingDB -  - ($)
08-25 10:40:46.814 15825 16181 W [-8]oneconnect[1.7.51.42_01][CORE]: KitchenServiceController.updatedDevicePlatformInfo - not supported deviceId : ($)2cecb532
08-25 10:40:46.814 15825 16181 I [-8]oneconnect[1.7.51.42_01][CORE]: ServiceBaseController.onLocationMessageReceived.MSG_DEVICE_PLATFORM_INFO_UPDATED - 510
08-25 10:40:46.814 15825 16181 W [-8]oneconnect[1.7.51.42_01][CORE]: PetCareController.updatedDevicePlatformInfo - not supported deviceId : ($)2cecb532
08-25 10:40:46.814 15825 13016 I [-8]oneconnect[1.7.51.42_01][CORE]: DeviceCloud.getRemoteRepresentation - uri : /capability/switch/main/0
```

what should I

Kwang-Hui, 2020-08-26:
@f34rdotcom @kolos @grobebar 
I'm sorry for your inconvenience.

Regarding "81-001" error, I'd like to suggest two kind of approach.

- remove all of existing test devices from your project --> register again.
or
- could you send us (stdk@samsung.com) detailed information ?
   1) actual device soft-ap ssid (e.g. `AD2IOTV10_E40AOf1006vheAM1NK` from @f34rdotcom 's log)
   2) screen capture of test device detailed information
   `Developer Workspace` --> `Project` --> `Test` --> `Test Devices` --> click one of your test device's serial number

[image]

grobebar, 2020-08-26:
@Kwang-Hui I just sent you the requested info via mail.

After looking more into the logs there is 1 interesting moment not sure if this helps:

    08-25 17:53:12.790 16706 16706 E [-8]oneconnect[1.7.51.42_01][MAIN_UI]: [EasySetup]GetCapiEncryptedKeyState.getCapiEncryptedKey - response body is empty
    08-25 17:53:12.790 16706 16706 I [-8]oneconnect[1.7.51.42_01][MAIN_UI]: [EasySetup]BaseStateMachine.showError - errorCode:81-001-MBL

jonggab-park, 2020-08-26:
> @Kwang-Hui I just sent you the requested info via mail.
>
> After looking more into the logs there is 1 interesting moment not sure if this helps:
>
> ```
> 08-25 17:53:12.790 16706 16706 E [...]

@grobebar 
If you are possible, could you test with new project??
After making new project on Developer Workspace, could you share the easysetup result??
If you get a fail, share **device SSID, serial nmber and the onboarding_config.json from Developer Workspace**, plz. ( stdk@samsung.com )
Thank you for your help.

grobebar, 2020-08-26:
This was actually with new project.
Inhave deleted everything from the Workspace (all projects, devices,
onboardings) and created each step new

Bartosz

bmccluskey, 2020-08-26:
Been having exactly the same problems and got a thread going on the communities. 
Have completly wiped all projects and rebuilt from scratch, project,devices,onboarding and I am getting
On the iphone I see the error is failToGetEasysetupBlob
On the Android it is 81-001

App versions
Android version 1.7.51.42
ios version 1.6.51-435

I can browse available wireless on phone and can see the ssid of device.  The phone apps never seem to connect or even attempt to connect.  Last week they appeared to connect as I could see it while monitoring the device and also there was a pop up on the phone and I could see the phone connected to the wifi of the device.

picam-jmartins, 2020-08-27:
I am also struggling with the same issue. 
Sent an email to stdk@samsung.com with the requested information.

Kwang-Hui, 2020-08-28:
Dear All,

We have found and fixed cloud side problem which occurred at some regions of us and eu.
Could you please try again?

Thanks for your issue reports and our apologies for the inconvenience.

Kwang-Hui Cho

grobebar, 2020-08-28:
Got a bit further now. with simply hitting retry it went to network selection but then got Error 38-012 (would be nice if these were more verbose  at least in the log file).

Got following error on the device monitor:

```
W (8525) [IoT]: iot_bsp_fs_read(297) > not found 'DeviceID'
W (8535) [IoT]: iot_security_storage_read(141) > id:10 not found
W (8545) [IoT]: _do_iot_main_command(576) > There are no reged data in NV

W (8555) [IoT]: _do_iot_main_command(583) > Some thing went wrong, got provisioning but no deviceId
I (8575) [IoT]: iot_device_cleanup(925) > start to erase device information
W (8665) [IoT]: iot_security_storage_remove(211) > id:10 not found
E (8675) [IoT]: iot_device_cleanup(937) > iot_device_cleanup: failed to erase device ID: -15
I (8675) [IoT]: event_handler(103) > Disconnect reason : 8
I (8775) [IoT]: event_handler(98) > SYSTEM_EVENT_STA_STOP

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

```

grobebar, 2020-08-28:
Interestingly hitting Retry without doing any changes produced a success. The device got connected with only following warnings:

```
W (2676) [IoT]: _do_iot_main_command(529) > There are no prov data in NV
W (2686) [IoT]: iot_security_storage_remove(211) > id:10 not found
```

```
W (278486) [IoT]: iot_bsp_fs_read(297) > not found 'MiscInfo'
W (278486) [IoT]: iot_security_storage_read(141) > id:11 not found
```

bmccluskey, 2020-08-28:
Ok - Generated new key, made QR, copied device json, re flashed, created new device in workspace.

Now the phone finds the device and I can join my wifi :slight_smile:
On the Android now gets to 99% although nothing seems to change on device from when app say about 50/60% then gives a 15-100 error
Below are last few messages from monitoring device.

    I (48824) [IoT]: _iot_net_tls_connect(160) > Loading the CA root certificate 2655@0x401098f0
    I (50854) [IoT]: iot_es_connect(1010) > MQTT connect success
    I (51124) [IoT]: _do_cmd_tout_check(913) > New state updated for 6

The iphone appears to do similar. My local wifi password now appear to be saved to device as i dont have to rekey password between join attempts even switching between phones.

Kwang-Hui, 2020-08-28:
> Interestingly hitting Retry without doing any changes produced a success. The device got connected with only following warnings:
>
> ```
> W (2676) [IoT]: _do_iot_main_command(529) > There are no prov [...]

These warning is pretty verbose. this could be come in case of test devices. let me consider to adjust it on next release.

Kwang-Hui, 2020-08-29:
> Ok - Generated new key, made QR, copied device json, re flashed, created new device in workspace.
>
> Now the phone finds the device and I can join my wifi :slight_smile:
> On the Android now gets to 99% [...]

Hi @bmccluskey ,
We we discussed on SmartThings community, it has been fixed. thank you.

Kwang-Hui, 2020-08-31:
This issue is closing.
If you have any further issues while on-boarding. Please open new issue.
thank you.

closed 2020-08-31 by Kwang-Hui

Neeraj-Parekh, 2025-04-09:
i am facing this same problem now with the latest version that it directs me to playstore saying - To add this category, SmartThings needs to be updated

jonggab-park, 2025-04-10:
> i am facing this same problem now with the latest version that it directs me to playstore saying - To add this category, SmartThings needs to be updated

Could you create new issue with more detail??