SmartThingsCommunity/st-device-sdk-c-ref #39: Feature Request while we all wait for 1.7.51...

f34rdotcom, 2020-08-10:
Add "Linux" build option.
Being able to build for an embedded device is good but sometimes we just want to run the code inside of a Raspberry Pi or small linux router. This would make this build environment even more flexible for building daemons on a Linux device.

YongsulOh, 2020-08-11:
Dear @f34rdotcom 

For linux environment, i think you need to check our main git : https://github.com/SmartThingsCommunity/st-device-sdk-c first
This ref. git is for esp32/esp8266/rtl8195/rtl8720c/rtl8721c/emw3166 now

On our main git , we support linux-like environment (please check 'Makefile' in there, now we support 'posix' style)
Unfortunately, we don't support "Raspberry Pi" now. (we are trying to support but not yet completed)
Please contribute us your new patch-set for "Raspberry Pi"

Kwang-Hui, 2020-08-12:
Hi @f34rdotcom ,
Thanks for raise feature request.
As @YongsulOh told, we are not support linux/raspberry-pi for now.
but we will take your feature request.

Kwang-Hui, 2022-12-22:
This issue remains as opened so long time. So closed

closed 2022-12-22 by Kwang-Hui