SmartThingsCommunity/st-device-sdk-c-ref #64: Kconfig not working for ESP8266

momo1, 2020-10-16:
Hi,

The Smartthings Kconfig file is not being picked up when using ESP8266 (works fine for ESP32). I copied the Kconfig from `iot-core/src` to my project's main/ so that menuconfig would pick it up. It's either that or define all the config variables manually, including "STDK_IOT_CORE" without which the source code won't even compile.

Kwang-Hui, 2020-10-20:
Hi,
Could you try like below?

```sh
$ python build.py apps/esp8266/switch_example/ menuconfig
```
`Component config` --> `SmartThings IoT Core`

[image]
[image]

Kwang-Hui, 2021-05-10:
This issue is closing now.
Please re-open if you still have this problem.

Thanks.

closed 2021-05-10 by Kwang-Hui