SmartThingsCommunity/st-device-sdk-c-ref #65: Example esp8266 code for sending temperature data

LucaSkywaIker, 2020-10-19:
[This ](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/blob/master/doc/getting_started.md) guide was very helpful at getting a smart light working with an esp8266, however there aren't examples on sending data (eg temperature sensor) to Smartthings.
In the light_example, there is functionality for a dummy dust sensor, however this makes use of header files such as caps_dustSensor.h and caps_dustSensor.c. Is there a place I could find premade temperatureMeasurement caps header files, or an example on how to do so?
Thank you for your assistance

f34rdotcom, 2020-10-19:
I use a few capabilities in my project maybe it will help. Still not well documented on some capabilities but most so far are simple. Also you can look from the results of the publish to the MQTT server in debugging if it was accepted and if not so far it has been a clear reply error from the server describing the issue.

https://github.com/nutechsoftware/AlarmDecoder-STSDK/tree/master/components/stsdk

LucaSkywaIker, 2020-10-19:
Thanks for your help!

Kwang-Hui, 2020-10-19:
> [This ](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/blob/master/doc/getting_started.md) guide was very helpful at getting a smart light working with an esp8266, however there aren't [...]

Hi @LucaSkywaIker ,
You can find most of standard capability handler files at https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/tree/master/apps/capability_sample (`caps_temperatureMeasurement.c` would be good for your purpose)

I think this document also helpful - https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/wiki/How-to-write-code-for-sending-device-event

Thanks.

LucaSkywaIker, 2020-10-19:
This is exactly what I needed, thank you.

closed 2020-10-19 by LucaSkywaIker