SmartThingsCommunity/st-device-sdk-c-ref #93: ESP8266 device onboarding error

jsp0929, 2021-06-30:
Hello,
I'm trying to build a device with eap8266 using smarttings SDK.
It always fails during onboarding process with error code 07-999.

I got information from st device SDK team that this unknown internal error(07-999) indicates that the WiFi connection between the esp8266 device and mobile phone is forcely closed. (I had sent them the log taken from my phone)

Here I open an issue on their request with the following serial terminal log.
I hope someone help with this.

--------------------- Terminal log -------------------------------------------------------------------------------------
Toolchain path: /home/jspark/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0
Python requirements from /home/jspark/st-device-sdk-c-ref/bsp/esp8266/requirements.txt are satisfied.
MONITOR

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 6508, room 16 
tail 12
chksum 0xe4
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x6d
ho 0 tail 12 room 4
load 0x3ffe8420, len 2832, room 12 
tail 4
chksum 0x52
csum 0x52
D (279) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
[0;32mI (300) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
D (304) nvs: nvs_open_from_partition backup_mac 0
D (306) nvs: nvs_get_str_or_blob backup_mac_data
D (311) nvs: nvs_close 1
[0;32mI (313) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
D (321) nvs: nvs_open_from_partition backup_mac 0
D (326) nvs: nvs_get_str_or_blob backup_mac_data
D (330) nvs: nvs_close 2
D (332) phy_init: loading PHY init data from application binary
D (338) nvs: nvs_open_from_partition phy 0
D (342) nvs: nvs_get_str_or_blob cal_data
D (346) nvs: nvs_get_str_or_blob dc_table
D (351) nvs: nvs_close 3
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
[0;32mI (413) phy_init: phy ver: 1163_0[0m
[0;32mI (416) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2[0m
D (420) wdt: Enable task watch dog panic, panic time parameter is 11
D (430) wdt: task watch dog trigger time parameter is 15
D (438) [IoT]: iot_nv_init(197) > [0;36m>>>HIT<<<[0;m
D (445) nvs: nvs_open_from_partition stdk 1
D (451) nvs: nvs_get_str_or_blob stnv
[0;32mI (457) [IoT]: _iot_bsp_fs_encryption(98) > 'nvs' is encrypted[0m
D (466) nvs: nvs_close 4
[0;32mI (479) [IoT]: _dump_device_info(706) > firmware_version: firmwareVersion_here[0m
[0;32mI (483) [IoT]: iot_bsp_wifi_init(170) > [esp8266] iot_bsp_wifi_init[0m
D (497) nvs: nvs_open_from_partition misc 1
D (499) nvs: nvs_open_from_partition nvs.net80211 1
D (506) nvs: nvs_get opmode 1
D (510) nvs: nvs_get_str_or_blob sta.ssid
D (516) nvs: nvs_get_str_or_blob sta.mac
D (521) nvs: nvs_get sta.authmode 1
D (526) nvs: nvs_get_str_or_blob sta.pswd
D (532) nvs: nvs_get_str_or_blob sta.pmk
D (538) nvs: nvs_get sta.chan 1
D (542) nvs: nvs_get auto.conn 1
D (547) nvs: nvs_get bssid.set 1
D (551) nvs: nvs_get_str_or_blob sta.bssid
D (557) nvs: nvs_get sta.lis_intval 2
D (562) nvs: nvs_get sta.phym 1
[... 164 lines cut ...]
D (36341) [IoT]: _iot_main_task(1143) > request step: 0

D (36349) [IoT]: iot_security_base64_encode_urlsafe(178) > src: 16@0x40108b98, dst: 25@0x40108bc8
D (36362) [IoT]: iot_security_base64_encode_urlsafe(192) > done: 24@0x40108bc8
D (36374) [IoT]: _iot_easysetup_gen_get_payload(156) > easysetup response for [0]
D (36384) [IoT]: _iot_easysetup_gen_get_payload(170) > payload: {"protocolVersion":"1.2.0","firmwareVersion":"firmwareVersion_here","hashedSn":"5jNJ2RQlX76_1TsNUq36GK_8xPqJTyYWPdOLB4sUJkg=","wifiSupportFrequency":
[0;32mI (36415) [IoT]: http_msg_handler(394) > get cmd[0] ok[0m
D (36422) [IoT]: iot_dump_log(346) > LOG : 30000548 00000024 0000018b 00000000
D (36433) [IoT]: http_msg_handler(450) > HTTP/1.1 200 OK

Server: SmartThings SDK Setup

Connection: keep-alive

Content-Type: application/json

Content-Length: 149

{"protocolVersion":"1.2.0","firmwareVersion":"firmwareVersion_here","hashedSn":"5jNJ2RQlX76_1TsNUq36GK_8xPqJTyYWPdOLB4sUJkg=","wifiSupportFrequency":
D (36563) event: SYSTEM_EVENT_AP_STADISCONNECTED, mac:5e:d0:bc:a7:3d:d4, aid:1
[0;32mI (36568) [IoT]: event_handler(153) > station:5e:d0:bc:a7:3d:d4 leave, AID=1[0m

----------------------------------------------------------------------------------------------------------------------------------------------

I quit the terminal log after onboarding error pops up.

Kwang-Hui, 2021-07-01:
Hello @jsp0929 ,

The log indicates your device generate incomplete json reponse like below
```
{"protocolVersion":"1.2.0","firmwareVersion":"firmwareVersion_here","hashedSn":"5jNJ2RQlX76_1TsNUq36GK_8xPqJTyYWPdOLB4sUJkg=","wifiSupportFrequency":
```
It should be complete json format including "iv" value like
```
{"protocolVersion":"1.2.0","firmwareVersion":"firmwareVersion_here","hashedSn":"5jNJ2RQlX76_1TsNUq36GK_8xPqJTyYWPdOLB4sUJkg=","wifiSupportFrequency":0,"iv":"xntx4EDX4gHHKtI75RvT-A=="}
```

I'd like to suggest you to add debugging log like below
https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/4367e87df5538aa46bf2cd5505abb30df80fbe91/src/easysetup/http/tcp/iot_easysetup_d2d_tcp.c#L429
```
     output_ptr = JSON_PRINT(root);
+  IOT_DEBUG("%s", output_ptr);
```

jsp0929, 2021-07-01:
Hi,
I got the missing part

{"protocolVersion":"1.2.0","firmwareVersion":"firmwareVersion_here","hashedSn":"5jNJ2RQlX76_1TsNUq36GK_8xPqJTyYWPdOLB4sUJkg=","wifiSupportFrequency":,"iv":"5szH6pY_cmaxO8rzP5dxJQ=="}

Kwang-Hui, 2021-07-01:
Newly added log indicates your device generating proper formatted message.
Do you still have same problem? If yes, please upload full the log including newly added line for further analysis.

And please try with turning of the Adaptive Wi-Fi 
Settings > Connections > Wi-Fi > More (:) > Advanced > Turn off "Switch to mobile data".

Thanks.

jsp0929, 2021-07-04:
It works!
I did two things. Rebuilt the device codes after deleting the build
folder(without any modification) and recreated the project on the st
developer workspace.
I'm not sure which one solved the problem.

Kwang-Hui, 2021-07-04:
Thanks.
If you have further question, Please open new issue.

closed 2021-07-04 by Kwang-Hui