SmartThingsCommunity/st-device-sdk-c-ref #110: "WIFI_STA_CONNECT_BIT event Timeout" error when finishing Onboarding on ESP8266

tatyanavolkova, 2021-11-16:
Hello everyone,

Not long ago a new error appeared, when trying to Direct-connect a ESP8266 board to SmartThings. I used a switch_example from this repo without any modifications. At first all goes well, the board is discovered and the phone asks which WiFi network to use. On the last step of Onboarding, the following error occurs:

E (134720) [IoT]: iot_bsp_wifi_set_mode(337) > WIFI_STA_CONNECT_BIT event Timeout
E (134726) [IoT]: _do_iot_main_command(564) > failed to set wifi_set_mode

After that, an erro with code 86-002 appears on smartphone screen.

See full log in the attach
[wifi_error.log](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/7550375/wifi_error.log)

jsp0929, 2021-11-17:
I got same thing with stdk v1.7.3. Hope this will be resolved soon!

stdk-scm, 2021-11-18:
Dear all,
Thank you for using, concerning and reporting this.
We're investigating this.
I'll update soon.

jy0209-kim, 2022-01-06:
Dear all,
This is resolved in v1.7.4.
There was an update to detect station disconnection during SoftAP mode and it made trouble in esp8266 due to the difference in performance between esp32.
Please try onboarding with v1.7.4 and let me know if you have trouble.
BR, Jiyoung.

closed 2022-01-20 by jy0209-kim