SmartThingsCommunity/st-device-sdk-c-ref #121: C++ Compilation

LucasianProfessor, 2022-12-04:
Hey, how would i go about including a c++ file in one of the /apps/esp8266/ examples? i tried messing with the makefile and the projects.mk file in the bsp folder but im struggling, help please

Kwang-Hui, 2022-12-21:
The build environment is depdent with esp8266 rtos sdk. 
I didn't try it by myself but could you refer[ this document](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-guides/build-system.html#optional-component-specific-variables) from ESP?

LucasianProfessor, 2022-12-22:
Thanks! I needed the C++ support for controlling WS2812 leds, which needs a C++ library like adafruit neopixel, neopixelbus, fastled, etc.

For anybody looking into this in the future (or you could add this to the wiki):
- renamed the main.c to main.cpp
- added CXXFLAGS += -fpermissive to my Makefile in the apps/esp8266/xxxxx_example directory (otherwise the compiler complains about unsafe castings)
- put extern "C" {} around the `#include "iot_uart_cli.h"` and `#include "iot_cli_cmd.h"`
- put extern "C" {} around app_main function
- included this library which has been patched to support the esp8266 rtos sdk: https://github.com/THKDev/NeoPixelBus_ESP8266_RTOS

Kwang-Hui, 2022-12-22:
Thanks for your update.
I think I can add `extern "C"`  to the header files.

Kwang-Hui, 2022-12-22:
Updated at [wiki](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/wiki)

closed 2022-12-22 by Kwang-Hui