SmartThingsCommunity/st-device-sdk-c-ref #122: ESP32 booting issue

cdragon-choi, 2023-01-01:
Hello,
I have pulled the st-device-sdk-c-ref at master branch and build apps/esp32/code_lab_example , flash it in ESP32-WROOM-32.
However, it fails to boot.
The error message is below.
Can i get some help?

**< ESP32-WROOM-32 boot message >**

```
ELF file SHA256: 54f4da97e7e6080f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6044
ho 0 tail 12 room 4
load:0x40078000,len:14464
load:0x40080400,len:4032
entry 0x40080650
E (31) qio_mode: Failed to set QIE bit, not enabling QIO mode
I (377) cpu_start: Pro cpu up.
I (378) cpu_start: Single core mode
I (386) cpu_start: Pro cpu start user code
I (386) cpu_start: cpu freq: 160000000
I (386) cpu_start: Application information:
I (390) cpu_start: Project name:     switch_example
I (396) cpu_start: App version:      v1.7.0-50-g6ec0fe5-dirty
I (402) cpu_start: Compile time:     Jan  1 2023 20:28:44
I (408) cpu_start: ELF file SHA256:  54f4da97e7e6080f...
I (414) cpu_start: ESP-IDF:          v4.3.1-3-g2c551b44e5
I (421) heap_init: Initializing. RAM available for dynamic allocation:
I (428) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (434) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (440) heap_init: At 3FFB7438 len 00028BC8 (162 KiB): DRAM
I (446) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (453) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (459) heap_init: At 40093F48 len 0000C0B8 (48 KiB): IRAM
I (466) spi_flash: detected chip: gd
I (469) spi_flash: flash io: qio
assertion "flash_ret == ESP_OK" failed: file "/home/myubuntu/st-device-sdk-c-ref/bsp/esp32/components/esp_system/startup.c", line 329, function: do_core_init

abort() was called at PC 0x40158033 on core 0

Backtrace:0x4008788f:0x3ffe3b10 0x40088009:0x3ffe3b30 0x4008e786:0x3ffe3b50 0x40158033:0x3ffe3bc0 0x400d344d:0x3ffe3bf0 0x400d34fe:0x3ffe3c10 0x40081201:0x3ffe3c50 0x4007913a:0x3ffe3c90 0x0

ELF file SHA256: 54f4da97e7e6080f

Rebooting...

```

junyoun-kim, 2023-03-03:
Could you try this example?
https://github.com/SmartThingsCommunity/st-device-sdk-c/tree/develop/example/esp32

junyoun-kim, 2024-11-27:
Hello, @cdragon-choi   Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim