SmartThingsCommunity/st-device-sdk-c-ref #126: ESP8266 switch_example SSID error

caleb-sav, 2023-02-26:
I ended up getting this to work, but I'm curious about what the problem might have been.

I was able to get the switch_example to flash to the ESP8266, but when I tried to connect to the device with my phone, the phone gave the error: "Security policy prevents use of unsecure Wi-Fi networks." I tried all the solutions I could find, but nothing worked. I had read that connecting to the wifi network manually can help the phone trust it, so I found where where the SSID and password were stored and added a line which printed these to the serial monitor. I was able to connect to the wifi, but the same error came back.

Finally, I started digging around for some of the errors that were in the serial monitor.
[image]
I found that the first one (`mbedtls_base64_decode = -0x002c`) has something to do with invalid characters. I noticed there were spaces in the SSID, which is derived from the device name, so I removed the spaces from the device name and re-downloaded the onboarding_info.json. Spaces disappeared from the SSID, but I still got the same error. I saw a comment in another issue to delete the build folder and rebuild and this caused everything to work!

My question is, why did deleting the build folder and starting from scratch fix the issue? Isn't everything in there overwritten when the project is rebuilt?

Finally, there's a plot twist. I put the spaces back in the device name just to see if that was really the problem, and this time, the SSID had underscores instead of spaces. Here's a screenshot of remembered networks to prove I'm not losing my mind.
[image]
So second question: Does anyone know how this could have happened? I'm something of a novice at this stuff, so I'd love to know why.

Kwang-Hui, 2023-05-31:
Could you check your `device_info.json` contains proper key-pair?
If you aren't set `CONFIG_STDK_IOT_CORE_SUPPORT_STNV_PARTITION`, then `iot-core` will use device key-pair from `device_info.json`

junyoun-kim, 2024-11-27:
Hello, @caleb-sav  Is there any more questions? We will close this issue if no more updates.

caleb-sav, 2024-11-28:
Sorry to ghost this thread @junyoun-kim ! You can close the issue, I'm not sure I could even remember what was going anymore.

closed 2024-11-28 by caleb-sav