SmartThingsCommunity/st-device-sdk-c-ref #132: fatal error:"st_dev.h" not found

jitensamsung, 2023-05-31:
I am getting this error when I am trying to flash esp8266 with example application. I have tried what was mentioned in [https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/issues/84](issue) but could not solve the problem. please help me ...

junyoun-kim, 2023-06-02:
@jitensamsung which branch you are working on? Did you try to use commands ./setup.sh esp8266 and ./build.sh esp8266 switch_example?

jitensamsung, 2023-06-06:
Yes I used both the commands and the shell files for setup and build executed. I also made changes in menuconfig. But when I am trying to flash, it says the header files from /iot-core folder are not found. 
@junyoun-kim

junyoun-kim, 2023-06-07:
@jitensamsung It's weird. We succeeded in building on develop branch. You are also working on latest develop branch right?

Could you let me know commit number you are currently building at? Also please share your full build log.

jitensamsung, 2023-06-07:
I am working on develop branch. 
`CC build/lwip/lwip/src/core/ipv6/dhcp6.o
CC build/lwip/lwip/src/core/ipv6/ip6_frag.o
CC build/lwip/lwip/src/netif/bridgeif.o
CC build/lwip/lwip/src/netif/lowpan6_common.o
CC build/lwip/lwip/src/netif/lowpan6_ble.o
CC build/lwip/lwip/src/netif/lowpan6.o
CC build/lwip/lwip/src/netif/zepif.o
CC build/lwip/lwip/src/netif/bridgeif_fdb.o
CC build/lwip/lwip/src/netif/ethernet.o
CC build/lwip/lwip/src/netif/slipif.o
CC build/lwip/port/esp8266/vfs_lwip.o
CC build/lwip/port/esp8266/freertos/sys_arch.o
CC build/lwip/port/esp8266/netif/dhcp_state.o
CC build/lwip/port/esp8266/netif/wlanif.o
CC build/lwip/port/esp8266/debug/lwip_debug.o
AR build/lwip/liblwip.a
CC build/main/caps_switch.o
In file included from C:/Users/jiten.m/Desktop/P2/msys32/home/jiten.m/esp/st-device-sdk-c-ref/apps/esp8266/switch_example/main/caps_switch.c:24:
C:/Users/jiten.m/Desktop/P2/msys32/home/jiten.m/esp/st-device-sdk-c-ref/apps/esp8266/switch_example/main/caps_switch.h:19:10: fatal error: caps/iot_caps_helper_switch.h: No such file or directory
 #include "caps/iot_caps_helper_switch.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/jiten.m/esp/st-device-sdk-c-ref/bsp/esp8266/make/component_wrapper.mk:292: caps_switch.o] Error 1
make: *** [/home/jiten.m/esp/st-device-sdk-c-ref/bsp/esp8266/make/project.mk:571: component-main-build] Error 2`
This is what I am getting.
@junyoun-kim

junyoun-kim, 2023-06-08:
@jitensamsung Looks like CONFIG_STDK_IOT_CORE is not set in sdkconfig. Could you check sdkconfig file under apps/esp8266/switch_example and how CONFIG_STDK_IOT_CORE is set?

jitensamsung, 2023-06-08:
This issue is solved. Thanks.

closed 2023-06-08 by jitensamsung