SmartThingsCommunity/st-device-sdk-c-ref #133: How to enable error logs in esp8266?

jitensamsung, 2023-06-08:
I am trying to setup a IOT device using esp8266. I've followed the steps mentioned in repo. I am able to flash but after flashing, the device is not connected to wifi and I can't even see any errors on monitor. Can someone please help me how to enable error logs ?
Thanks

junyoun-kim, 2023-06-08:
@jitensamsung Did you try using `./build.py esp8266 switch_example monitor` ?

Also if you have esp32, could you try with it?
If you are starter, we recommend esp32 chip rather than esp8266. It has more enhanced features and memory size.

jitensamsung, 2023-06-13:
I don't have esp32. Can you please tell me what settings are required in menuconfig other than setting port,wifi ssid and password. Because I am not able to connect the board to wifi and even error is not coming on monitor so that I can debug. I think I a missing something in menuconfig. Can you please tell what all settings need to be done in menuconfig for switch_example on esp8266 board.
@junyoun-kim

junyoun-kim, 2023-06-13:
Did you follow https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/blob/develop/doc/getting_started.md document?

Device get wifi info through SmartThings app. So you can't set manually wifi ssid and password.

jitensamsung, 2023-06-15:
I think we have a option for giving ssid and password in menuconfig.

junyoun-kim, 2023-06-19:
Those settings are not applied our example. Device can wifi information only by SmartThings app.

jitensamsung, 2023-06-19:
But if my device is not detected by app, how will it get the wifi info?

junyoun-kim, 2023-06-19:
You device should be detected by app. Wifi info should be transferred during process of registering a device on SmartThings server.

If the device can get wifi info without registering through app, it's just a device connected to internet not SmartThings registered device.

jitensamsung, 2023-06-21:
Th device is getting detected but it is not fully connected. I am continuing the same issue here ( #134 ) . So closing this issue.

closed 2023-06-21 by jitensamsung