SmartThingsCommunity/st-device-sdk-c-ref #134: ESP8266 switch_example

jitensamsung, 2023-06-16:
I am working on switch_example on esp8266 using the code from this repo. Have followed all the steps mentioned in this repo and espressif website. Also I have run the example from ESP_RTOS_STDK so that my enviorment check is done. But now when I am trying with switch_example, I am stuck at this point. I couldn't figure out whats the problem and my device is not connected to wifi and smarthtings app also not recognizing the device.
```
I (353) [IoT]: _dump_device_info(747) > firmware_version: switch_example_001
I (356) [IoT]: iot_bsp_wifi_init(193) > [esp8266] iot_bsp_wifi_init
I (362) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (375) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (450) phy_init: phy ver: 1163_0
I (475) [IoT]: iot_bsp_wifi_init(231) > [esp8266] iot_bsp_wifi_init done
I (479) [IoT]: st_conn_init(1134) > stdk_version : 1.8.7
I (481) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0
I (495) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
��*�P*V��ET����T��K��u�U+UZ�Z�T��T
��*�P*V��ET����T��K��u�U+UZ�Z�T��T����Eu��T�QQ��jPU���u�����ZT�(UE����Q�UEu�����*�B����B�u��E+���T)UZV�R�Eu����ŪQB���U�*�u���ZT�(E����QT�E�j*QT��I (355) [IoT]: _dump
                                  ����Eu��T�QQ��jPU���u�����ZT�(UE����Q�UEu�����*�B����B�u��E+���T)UZV�R�Eu����ŪQB���U�*�u���ZT�(E����QT�E�j*QT��I (355) [IoT]: _dump_device_info(747) > firmware_version:
 switch_example_001
I (358) [IoT]: iot_bsp_wifi_init(193) > [esp8266] iot_bsp_wifi_init
I (364) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (377) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (453) phy_init: phy ver: 1163_0
I (477) [IoT]: iot_bsp_wifi_init(231) > [esp8266] iot_bsp_wifi_init done
I (481) [IoT]: st_conn_init(1134) > stdk_version : 1.8.7
I (484) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0
I (498) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```
also when I have connected the LED on board, the switch thing is working. Please help me...

junyoun-kim, 2023-06-19:
I think GPIO config went wrong. Could you check device_control.h file in switch_example?

There are GPIO pint number setting like below. You can modify GPIO Pin settings according to your device kit.

`#if defined(CONFIG_TARGET_WITTY_CLOUD)

#define GPIO_INPUT_BUTTON 4

#define GPIO_OUTPUT_MAINLED 15
#define GPIO_OUTPUT_MAINLED_0 16 /* use as ground */

#define GPIO_OUTPUT_NOUSE1 12
#define GPIO_OUTPUT_NOUSE2 13

#elif defined(CONFIG_TARGET_ESP8266_DEVKITC_V1)

#define GPIO_INPUT_BUTTON 0

#define GPIO_OUTPUT_MAINLED 14
#define GPIO_OUTPUT_MAINLED_0 15 /* use as ground */

#define GPIO_OUTPUT_NOUSE1 12
#define GPIO_OUTPUT_NOUSE2 13

#else //default

#define GPIO_OUTPUT_NOTIFICATION_LED 2
#define GPIO_INPUT_BUTTON 5

#define GPIO_OUTPUT_MAINLED 16
#define GPIO_OUTPUT_MAINLED_0 13 /* use as ground */

#define GPIO_OUTPUT_NOUSE1 14
#define GPIO_OUTPUT_NOUSE2 12

#endif`

jitensamsung, 2023-06-19:
Actually I need only one button and one LED pin. Button->5 and LED->14,13. So I have removed the extra pins. And there was a issue with GPIO pins. I settled it but now I am stuck at this point.
```
I (353) [IoT]: _dump_device_info(747) > firmware_version: switch_example_001
I (357) [IoT]: iot_bsp_wifi_init(193) > [esp8266] iot_bsp_wifi_init
I (363) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (376) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (451) phy_init: phy ver: 1163_0
I (476) [IoT]: iot_bsp_wifi_init(231) > [esp8266] iot_bsp_wifi_init done
I (480) [IoT]: st_conn_init(1134) > stdk_version : 1.8.7
I (482) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (496) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (511) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
iot_gpio_init done
I (2522) [IoT]: st_conn_start(1222) > st_conn_start start (pin)
I (2536) [IoT]: _do_iot_main_command(586) > curr_main_cmd:0, curr_main_state:0
I (2539) [IoT]: _do_state_updating(435) > current state 0, new state 2
I (2545) [IoT]: iot_bsp_wifi_set_mode(249) > iot_bsp_wifi_set_mode = 1
I (2558) [IoT]: iot_bsp_wifi_set_mode(280) > WiFi Station Started
I (2569) [IoT]: st_conn_start(1277) > st_conn_start done (0)
end_st_conn_startI (4631) [IoT]: event_handler(182) > event_handler = 1
I (4689) [IoT]: iot_easysetup_create_ssid(91) > >> NAME[QD5f] <<
I (4692) [IoT]: iot_bsp_wifi_set_mode(249) > iot_bsp_wifi_set_mode = 3
I (4700) [IoT]: event_handler(150) > SYSTEM_EVENT_AP_START
I (6815) [IoT]: event_handler(155) > SYSTEM_EVENT_AP_STOP
I (6819) [IoT]: event_handler(150) > SYSTEM_EVENT_AP_START
I (6823) [IoT]: iot_bsp_wifi_set_mode(410) > AP Mode Started
I (6829) [IoT]: iot_easysetup_init(492) > IOT_STATE_PROV_ES_START
I (6839) [IoT]: es_http_init(159) > http tcp init!!
I (6850) [IoT]: iot_easysetup_init(528) > IOT_STATE_PROV_ES_INIT_DONE
I (6859) [IoT]: _do_state_updating(533) > Current timeout : 300000 for 0
status: 2, stat: 1
I (6873) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x102
I (306974) [IoT]: _do_cmd_tout_check(822) > Go into OOB sleep mode
I (306977) [IoT]: http_cleanup_all_connection(39) > close listen socket
I (306983) [IoT]: es_http_deinit(184) > http tcp deinit complete!
I (306992) [IoT]: iot_easysetup_deinit(564) > IOT_STATE_PROV_ES_DONE
I (307003) [IoT]: iot_bsp_wifi_set_mode(249) > iot_bsp_wifi_set_mode = 0
I (307016) [IoT]: event_handler(155) > SYSTEM_EVENT_AP_STOP
I (307025) [IoT]: event_handler(111) > SYSTEM_EVENT_STA_STOP
I (307035) [IoT]: _do_iot_main_command(586) > curr_main_cmd:0, curr_main_state:2
I (307046) [IoT]: _do_state_updating(435) > current state 2, new state 1
I (307057) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x102
```
LED is glowing on board at GPIO 14,15. But this device is not recognized by smartthings app yet. Also the code gets stuck at this point. Please help.

junyoun-kim, 2023-06-20:
Did you enable developer mode in smartthings app? Also could you check there is wifi advertising SSID something like NAME_E4***QD5f.

jitensamsung, 2023-06-20:
Yes I have enabled developer mode. Where to check this wifi SSID?
@junyoun-kim

junyoun-kim, 2023-06-20:
Did you also change onboarding_config.json with yours? Looks like you use default empty onboarding_config.json.

Your onboarding_config.json can be downloaded from your Project on DevWS.

jitensamsung, 2023-06-20:
Yes I did change onboarding config and device info

junyoun-kim, 2023-06-20:
it's wired. Then your onboardingId is "NAME"? In below log, looks like onboardingId set as "NAME".

`I (4689) [IoT]: iot_easysetup_create_ssid(91) > >> NAME[QD5f] <<`

Could you share your onboarding_config.json? Also could you clean build and rebuild it?

jitensamsung, 2023-06-20:
```
{
  "onboardingConfig": {
    "deviceOnboardingId": "CodeLabExampl",
    "mnId": "0AF8",
    "setupId": "002",
    "vid": "ST_224789ba-dffd-45fc-9dc8-57b93b0ec77b",
    "deviceTypeId": "Switch",
    "ownershipValidationTypes": [
      "BUTTON"
    ],
    "identityType": "ED25519",
    "deviceIntegrationProfileKey": {
      "id": "ab2134ff-b6f4-475b-b32f-947beb1fa7cc",
      "majorVersion": 0,
      "minorVersion": 1
    },
    "productId": "569e5b9f-6da9-42dc-8cde-d70ae555502b"
  }
}
``` 
This is my onboarding.json

junyoun-kim, 2023-06-20:
Then please clean and try to rebuild it after replacing json files.

```
python build.py apps/esp8266/switch_example/ clean
python build.py apps/esp8266/switch_example/ flash monitor
```

jitensamsung, 2023-06-20:
My device is detected by smartthings app. But it is not fully connected. It gives an error after 34% saying
```
 "There was a problem verifying your device. Unplug the device and plug it in again, then try adding the device again."
 Error code: 38-007
```
And giving this error on monitor
```
I (6852) [IoT]: iot_easysetup_init(528) > IOT_STATE_PROV_ES_INIT_DONE
I (6861) [IoT]: _do_state_updating(533) > Current timeout : 300000 for 0
status: 2, stat: 1
I (6875) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x102
I (9027) [IoT]: event_handler(162) > station:b0:72:bf:86:01:87 join, AID=1
I (9031) [IoT]: _iot_easysetup_wifi_event_cb(91) > Station joined to SoftAP
I (9475) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (9480) [IoT]: event_handler(182) > event_handler = 18
status: 2, stat: 4
I (10015) [IoT]: _iot_easysetup_gen_get_payload(174) > waiting.. response for [0]
**E (10029) [IoT]: iot_misc_info_load(1161) > Can't get misc_info from NV(-14)**
I (10044) [IoT]: http_msg_handler(419) > get cmd[0] ok
I (10124) [IoT]: _iot_easysetup_gen_post_payload(284) > waiting.. response for [1]
I (10962) [IoT]: _es_keyinfo_handler(564) > master secret generation success
I (10980) [IoT]: http_msg_handler(395) > post cmd[1] ok
I (11021) [IoT]: event_handler(172) > station:b0:72:bf:86:01:87 leave, AID=1
I (11025) [IoT]: _iot_easysetup_wifi_event_cb(95) > Station left from SoftAP
**E (11031) [IoT]: _iot_easysetup_wifi_event_cb(99) > mock : signaling restart onboarding 99
E (35255) [IoT]: http_packet_read(96) > recv failed: errno 113
E (35258) [IoT]: process_accepted_connection(65) > failed to read http packet -561**
```
and this also
```
**E (126477) [IoT]: _iot_easysetup_wifi_event_cb(99) > mock : signaling restart onboarding 99**
I (129681) [IoT]: event_handler(162) > station:b0:72:bf:86:01:87 join, AID=1
I (129685) [IoT]: _iot_easysetup_wifi_event_cb(91) > Station joined to SoftAP
I (129692) [IoT]: event_handler(172) > station:b0:72:bf:86:01:87 leave, AID=1
I (129704) [IoT]: _iot_easysetup_wifi_event_cb(95) > Station left from SoftAP
**E (129715) [IoT]: _iot_easysetup_wifi_event_cb(99) > mock : signaling restart onboarding 99**
I (141285) [IoT]: event_handler(162) > station:b0:72:bf:86:01:87 join, AID=1
I (141289) [IoT]: _iot_easysetup_wifi_event_cb(91) > Station joined to SoftAP
I (141426) [IoT]: event_handler(172) > station:b0:72:bf:86:01:87 leave, AID=1
I (141429) [IoT]: _iot_easysetup_wifi_event_cb(95) > Station left from SoftAP
```
Errors are highlighted in bold.
@junyoun-kim

junyoun-kim, 2023-06-21:
Are you using IOS or Android? And your ST app looks like old version.

Could you update latest ST app and try again?

jitensamsung, 2023-06-21:
It says,
```
This device isn't supported in your Samsung account's country or region 
Error code: 07-008
```
on the latest version(1.8.01.22) downloaded from play store.

junyoun-kim, 2023-06-21:
Did you enable developer mode after reinstall?

Also could you check https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/blob/develop/doc/getting_started.md#add-a-product-info

At above, you can add availability country in product info. You need to add your samsung account country at that.

jitensamsung, 2023-06-21:
Yes I enabled developer mode. Atleast on the older version, the connection was starting but new versions are not allowing to connect the device @junyoun-kim

junyoun-kim, 2023-06-21:
Then how about country availability setting?

jitensamsung, 2023-06-21:
The site is not currently supporting. But as far as I know, India and US regions are set as available there. But I will check as soon as the site opens

jitensamsung, 2023-06-21:
Yes India and USA are the regions added on Developers workspace.
[image]

junyoun-kim, 2023-06-21:
Hmm.. Could you share mobile log? We need ST app log to identify issue.

jitensamsung, 2023-06-21:
How can I share them?

junyoun-kim, 2023-06-21:
On phone number, press *#9900#.

Press "Run dumpstate/logcat" and after that, "Copy to sdcard". And then you can get dumpstate file under "Log" folder.

junyoun-kim, 2023-06-23:
After unzip the log you sent, dumpstate file in it was 0 byte. Could you check your dumpstate file?

jitensamsung, 2023-06-26:
@junyoun-kim I have shared the file now. Can you please check it?

jitensamsung, 2023-07-03:
The connection process goes till 99% but then this error is coming.
```
I (132985) [IoT]: _iot_net_tls_connect(418) > Loading the CA root certificate 1317@0x40109d28
D (133027) [IoT]: _iot_net_tls_connect(436) > Connecting to mqtt-regional-apnortheast2.api.smartthings.com:8883
E (139536) [IoT]: _iot_net_tls_connect(443) > mbedtls_net_connect = -0x0044
E (139539) [IoT]: _iot_mqtt_connect_net(798) > MQTT net connection failed
E (139545) [IoT]: _iot_es_mqtt_connect(939) > _iot_es_mqtt_connect error(-1)
E (139557) [IoT]: iot_es_connect(1194) > failed to connect
E (139566) [IoT]: _do_iot_main_command(613) > failed to iot_es_connect for registration
```
@junyoun-kim please help...

junyoun-kim, 2023-07-03:
Please check your email

closed 2024-11-27 by junyoun-kim