SmartThingsCommunity/st-device-sdk-c-ref #137: esp32-c3 issue when testing device

LUSAN54, 2023-12-16:
1.            A detailed description of the issue

I´m trying to test my first directly connected device. And in the testing process with the mobile phone an error occurs. The error number that appears on the mobile is 86-002. Look at the photo of the mobile screen that i attached.

     I’m following the tutorial described in Direct Connected Devices thata are on the Samsung Developers site.

     I’m using A ESP32C3 DevKit as device.

     I am comiling and loading in the decive the example in st-device-sdk-c-ref/apps/esp32c3/switch_example.

     I have followed the enteri process as indicated.

     When I start the testing process, I see my device on the mobile phone.

     I select my device, then the mobile asks me for the Wi-Fi network to connect the device to.

     I select the wifi network and everythin seems to be fine but after a while the phone displays the screen with error 86-002.

     I have attached the log file that shows the device. In the file I see that at one point somethings is not right in the following lines:

 

I (59377) [IoT]: _iot_net_tls_connect(418) > Loading the CA root certificate 1317@0x3fcaffc4
I (60147) [IoT]: iot_es_connect(1196) > MQTT connect success
I (60247) [IoT]: _check_connection_response(79) > Connection response payload {"target":"STDK8ZNT4qNqnYkZ","currentTime":1702376425,"event":"expired.jwt"}
I (60247) [IoT]: _check_connection_response(92) > Set SNTP with current time 1702376425
W (60257) [IoT]: iot_es_connect(1227) > GG connection fail
I (60267) [IoT]: st_mqtt_disconnect(1434) > mqtt disconnect 0
I (60277) [IoT]: _iot_net_show_status(92) > [1702376425] network socket status: sockfd 54 readable 0 writable 4194304 sock_err 0 errno 34
E (60287) [IoT]: _do_iot_main_command(613) > failed to iot_es_connect for registration
E (60297) [IoT]: _iot_main_task(909) > failed handle cmd (1): -201

 

SmartThings app version (Menu > Settings > About SmartThings)
1.8.10.21

 

 

Mobile model number and OS versión
SAMSUNG GALAXY S22 ANDROID 12

Location from where you are contacting usc
SPAIN
[registro.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/13692399/registro.txt)
[image]

junyoun-kim, 2023-12-18:
Thank you for having interest in our SDK.

It looks like there was a bugfix in iot-core sub-module. We've just update latest one in develop branch( #138 ).

Could you rebuild and test it again with latest version?

LUSAN54, 2023-12-18:
Thank you for the support.
With this version 1.13.18 I have been able to test my device correctly.
Thanks.

Enviado el: lunes, 18 de diciembre de 2023 8:04
Para: SmartThingsCommunity/st-device-sdk-c-ref ***@***.***>
CC: LUIS ANTONIO SANCHEZ CANO ***@***.***>; Author ***@***.***>

junyoun-kim, 2024-11-27:
Hello, @LUSAN54  Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim