SmartThingsCommunity/st-device-sdk-c-ref #139: esp32 build error (switch example)

asurada92, 2023-12-20:
hello,
i’m trying to get setup for SDK for esp32 board
After all setting( setup esd-if, etc…)
when i execute build.py (ST SDK),

`Toolchain path: /home/yhpark/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-97-gc752ad5
Expected to see version: crosstool-ng-1.22.0-96-g2852398
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-97-gc752ad5
Expected to see version: crosstool-ng-1.22.0-96-g2852398
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/yhpark/st-device-sdk-c-ref/bsp/esp32_v3.3/requirements.txt are satisfied.

App "switch_example" version: v1.8.7-2-ga8226c6-dirty
CC /home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/build/main/device_control.o
/home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/main/device_control.c: In function 'get_button_event':
/home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/main/device_control.c:52:20: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'uint32_t {aka unsigned int}' [-Werror=format=]
             printf("Button event, val: %ld, tick: %lu\n", gpio_level, (uint32_t)xTaskGetTickCount());
                    ^
/home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/main/device_control.c:52:20: error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Werror=format=]
/home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/main/device_control.c:52:20: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'uint32_t {aka unsigned int}' [-Werror=format=]
/home/yhpark/st-device-sdk-c-ref/apps/esp32/switch_example/main/device_control.c:52:20: error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Werror=format=]
cc1: some warnings being treated as errors
make[1]: *** [/home/yhpark/st-device-sdk-c-ref/bsp/esp32_v3.3/make/component_wrapper.mk:290: device_control.o] Error 1
make: *** [/home/yhpark/st-device-sdk-c-ref/bsp/esp32_v3.3/make/project.mk:552: component-main-build] Error 2`

Error that related with data format has occured…
how can i fix it?

junyoun-kim, 2023-12-20:
Hello,

Could you try this example?(https://github.com/SmartThingsCommunity/st-device-sdk-c/tree/develop/example/esp32)

There is README file, so you can follow the instruction.

asurada92, 2024-01-07:
> Hello,
> 
> Could you try this example?(https://github.com/SmartThingsCommunity/st-device-sdk-c/tree/develop/example/esp32)
> 
> There is README file, so you can follow the instruction.

Now I can use "idf.py build" and I can monitor the chip(esp32) state through idf.py flash
but how can I connect to smart things ?
Ofcoures, I already set the "device_info.json" and "onboarding_config.json"
and I already register my device on "smartthings.developer.samsung.com/"
But after I make qrcode
when I use the qrcode (in add device at smartthings app), the err is occured ("지원되지 않는 QR코드 이 QR코드로는 기기를 smartthings에 추가할 수 없습니다....)
Ofcoures I already activate developer mode (in iOS)

what's the problem? can you help me?

junyoun-kim, 2024-01-08:
@asurada92 Could you let me know how you made qr code? and Could you share your qr code image?

asurada92, 2024-01-08:
> @asurada92 Could you let me know how you made qr code? and Could you share your qr code image?

I made qr code referencing this page "https://github.com/SmartThingsCommunity/st-device-sdk-c/blob/develop/doc/getting_started.md"

Technically speaking, I use the python code below

==========================================
import qrcode

mnid = 'FFFF' # "FFFF" is an example. you should replace it with yours
onboardingId = '111' # "111" is an example. you should replace it with yours
serialNumber = 'STDKtest0001' # "STDKtest0001" is an example. you should replace it with yours
qrUrl = 'https://qr.samsungiots.com/?m=' + mnid + '&s=' + onboardingId + '&r=' + serialNumber
qrcode.QRCode(box_size=10, border=4)
img = qrcode.make(qrUrl)
img.save(serialNumber + '.png')
===========================================
here is the qr code image

[image]

junyoun-kim, 2024-01-08:
@asurada92 When I parsed the qrcode, it looks like below,

https://qr.samsungiots.com/?m=f3WE&s=onboarding&r=STDKrF6jsWAKcys9

And I think you put "onboarding" string instead of your onboardingId. Could you check this?

asurada92, 2024-01-08:
> @asurada92 When I parsed the qrcode, it looks like below,
> 
> https://qr.samsungiots.com/?m=f3WE&s=onboarding&r=STDKrF6jsWAKcys9
> 
> And I think you put "onboarding" string instead of your onboardingId. Could you check this?

yes, you're right 
however,I already know that the onboarding string is on the qr-code
because the qr-generation code of mine is below

[image]

I put the 'onboarding' string referencing "onboardingConfig.json" file like below
[image]

And I know that this onboardingconfig.json file can be downloaded from developer work space

I guess I can change onboarding ID through changing "onboarding display name"
[image]

isn't it? I guess this can be string

thank you for your co-operation

junyoun-kim, 2024-01-08:
@asurada92 oh, Sorry for confusing you. you should put "setupId" in "onboardingId" value. So your case should be like below.

In python script,
....
onboardingId = '001'
....

We will change "onboardingId" to "setupId" in document to prevent confusion.

asurada92, 2024-01-08:
> @asurada92 oh, Sorry for confusing you. you should put "setupId" in "onboardingId" value. So your case should be like below.
> 
> In python script, .... onboardingId = '001' ....
> 
> We will change "onboardingId" to "setupId" in document to prevent confusion.

Hello junyoun-kim,
I changed onboardingid to -> 001 but err still occured

here is device_info.json code, onboarding_config.json code
[image]

And this is qr-code that re-generated
[image]

junyoun-kim, 2024-01-09:
@asurada92 Is this same err? Could you let me know which err you are facing now?

asurada92, 2024-01-09:
> @asurada92 Is this same err? Could you let me know which err you are facing now?

지원되지 않는 QR코드 이 QR코드로는 기기를 smartthings에 추가할 수 없습니다....)
Above err occured

junyoun-kim, 2024-01-09:
@asurada92 Could you check your iOS ST app's samsung account is same as DevWS account? And also could you check enabling developer mode in ST app?

asurada92, 2024-01-10:
[image]

[image]

the two account is same

and 

[image]

I also turned on the developer mode....

junyoun-kim, 2024-01-11:
@asurada92 We found out there is problem in iOS ST APP for testing in developer mode. Sorry for letting you know this. We will let you know when fix is published.

Meanwhile you can try it with Android ST app.

junyoun-kim, 2024-01-15:
@asurada92 iOS ST APP issue patch will be applied next app update.

asurada92, 2024-01-17:
> @asurada92 iOS ST APP issue patch will be applied next app update.

When does the update will be on?

junyoun-kim, 2024-01-18:
@asurada92 That will be Jan. 22.

junyoun-kim, 2024-11-27:
Hello, @asurada92  Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim