SmartThingsCommunity/st-device-sdk-c-ref #142: esp8266 build error

mengxw8, 2024-01-11:
在使用官方推荐的`python build.py esp8266 switch_example` 进行编译的时候会有如下的错误：
`Generating esp8266.project.ld
LD build/switch_example.elf
/mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o):(.literal.F88ef4f295d2+0x0): undefined reference to `iot_bsp_system_get_uniqueid'
/mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o): in function `F88ef4f295d2':
/home/shkim/project/things/stdk/github-sec/st-device-sdk-c-ref/bsp/esp8266/components/iot_security_ss/iot_security_ss.c:55: undefined reference to `iot_bsp_system_get_uniqueid'
/mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o): in function `F75695623470':
/home/shkim/project/things/stdk/github-sec/st-device-sdk-c-ref/bsp/esp8266/components/iot_security_ss/iot_security_ss.c:91: undefined reference to `iot_bsp_system_get_uniqueid'
collect2: error: ld returned 1 exit status
make: *** [/mnt/f/c/st-device-sdk-c-ref/bsp/esp8266/make/project.mk:510: /mnt/f/c/st-device-sdk-c-ref/apps/esp8266/switch_example/build/switch_example.elf] Error 1`

经过排查发现是因为在iot-core库中的一次提交feat : Remove unused bsp porting function移除掉了iot_bsp_system_get_uniqueid，此次的提交id为b19fbbe71d5305749aabc4127fec8c9838696b8e 希望修复一下编译的问题

谢谢

junyoun-kim, 2024-01-11:
Hello, Could you make it in English?

mengxw8, 2024-01-11:
您好，这是自动回复！您的邮件我已经收到，我会在第一时间进行处理。—————————————————————————————Thank you. Your email is received and will be handled  as soon as possible.

mengxw8, 2024-01-11:
When using the official recommended commands for compilation, there may be the following errors(python build.py esp8266 switch_example):
Generating esp8266.project.ld LD build/switch_example.elf /mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o):(.literal.F88ef4f295d2+0x0): undefined reference to iot_bsp_system_get_uniqueid'
/mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o): in function F88ef4f295d2': /home/shkim/project/things/stdk/github-sec/st-device-sdk-c-ref/bsp/esp8266/components/iot_security_ss/iot_security_ss.c:55: undefined reference to iot_bsp_system_get_uniqueid'
/mnt/e/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /mnt/f/c/st-device-sdk-c-ref/iot-core/src/security/backend/software/lib/esp/libiot_security_ss.a(iot_security_ss.o): in function F75695623470': /home/shkim/project/things/stdk/github-sec/st-device-sdk-c-ref/bsp/esp8266/components/iot_security_ss/iot_security_ss.c:91: undefined reference to iot_bsp_system_get_uniqueid'
collect2: error: ld returned 1 exit status
make: *** [/mnt/f/c/st-device-sdk-c-ref/bsp/esp8266/make/project.mk:510: /mnt/f/c/st-device-sdk-c-ref/apps/esp8266/switch_example/build/switch_example.elf] Error 1`

I suspect it was a submission from the iot-core code repository ，“feat : Remove unused bsp porting function”，Removed iot_bsp_system_get_uniqueid function，Submission node：b19fbbe71d5305749aabc4127fec8c9838696b8e 

Hope to solve the compilation problem

thank you.

The above content is from the machine translation, I hope you can understand。

奈何桥上摆地摊
***@***.***

junyoun-kim, 2024-01-12:
@mengxw8 We've fixed the issue with #143 . Could you setup and build with latest commit?

mengxw8, 2024-01-15:
I used your newly submitted code to build successfully, thank you for your fix。 @junyoun-kim

closed 2024-11-27 by junyoun-kim