SmartThingsCommunity/st-device-sdk-c-ref #147: Issues OTA DEMO connecting with local server

LUSAN54, 2024-02-21:
I´m usign the example of the SmartThings sdk OTA-DEMO.
The RaspberryPi is being used as an ubuntu machine as a local OTA server.
I have followed all the instructions to generate the necesary certificates and keys accordin to the OTA DEMO server.
At the end i have tested the server access from a computer with the command :
curl --cacert ./root.crt "https://192.168.67.35:4443"
And the result has been 👍  
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN" "http://www.w3.org/TR/html4/strict.dtd">
<html>
<head>
<meta http-equiv="Content-Type" content="text/html; charset=iso8859-1">
<title>Directory listing for /</title>
</head>
<body>
<h1>Directory listing for /</h1>
<hr>
<ul>
<li><a href="signed_switch_example.bin">signed_switch_example.bin</a></li>
<li><a href="version_info.json">version_info.json</a></li>
</ul>
<hr>
</body>
</html>

The i programmed the ESP32 device with de OTA-DEMO program and the device works correctly and i can send ON -OFF commands form the SmarThigns app.
**The problem is that de device can´t connect to the server. You can see below the logs register.**
i´m stuck and i don´t know where to go next in my produc devolopment:
lusan@DESKTOP-CL4R1HL:~/st-device-sdk-c-ref$ python build.py apps/esp32/ota_demo monitor
/bin/sh: 22: ./export.sh: [[: not found
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /home/lusan/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/requirements/requirements.core.txt
Python being checked: /home/lusan/.espressif/python_env/idf5.0_py3.10_env/bin/python
/home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements are satisfied.
Updated PATH variable:
  /home/lusan/st-device-sdk-c-ref/bsp/esp32/components/espcoredump:/home/lusan/st-device-sdk-c-ref/bsp/esp32/components/partition_table:/home/lusan/st-device-sdk-c-ref/bsp/esp32/components/app_update:/home/lusan/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/lusan/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/lusan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/lusan/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/lusan/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/lusan/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/lusan/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/lusan/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:/home/lusan/.espressif/python_env/idf5.0_py3.10_env/bin:/home/lusan/st-device-sdk-c-ref/bsp/esp32/tools:/home/lusan/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/WindowsApps/CanonicalGroupLimited.Ubuntu22.04LTS_2204.3.63.0_x64__79rhkp1fndgsc:/mnt/c/Program Files/Common Files/Oracle/Java/javapath:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/usbipd-win/:/mnt/c/Users/LUSAN.DINUY/AppData/Local/Programs/Python/Python312/Scripts/:/mnt/c/Users/LUSAN.DINUY/AppData/Local/Programs/Python/Python312/:/mnt/c/Users/LUSAN.DINUY/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/LUSAN.DINUY/AppData/Local/Programs/Microsoft VS Code/bin:/snap/bin

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of riscv32-esp-elf, riscv32-esp-elf-gdb, xtensa-esp32-elf, esp32ulp-elf, esp-rom-elfs, xtensa-esp32s2-elf, xtensa-esp32s3-elf, xtensa-esp-elf-gdb, esp32s2ulp-elf, openocd-esp32 use command 'python /home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools. Use option 'python3 /home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

/home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running idf_monitor in directory /home/lusan/st-device-sdk-c-ref/apps/esp32/ota_demo
Executing "/home/lusan/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/lusan/st-device-sdk-c-ref/apps/esp32/ota_demo/build/ota_demo.elf -m '/home/lusan/.espressif/python_env/idf5.0_py3.10_env/bin/python' '/home/lusan/st-device-sdk-c-ref/bsp/esp32/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[... 117 lines cut ...]
I (8811) [IoT]: iot_cap_call_init_cb(1341) > Call init_cb for switch capability
I (8821) [IoT]: st_cap_send_attr(613) > publish event, topic : /v1/deviceEvents/6543a7c7-2d1c-4e86-8675-ba35fea2a1e9, payload :
{"deviceEvents":[{"component":"main","capability":"switch","attribute":"switch","value":"on","providerData":{"sequenceNumber":1,"timestamp":"1708535020023"}}]}
Sequence number return : 1
I (8851) [IoT]: iot_cap_call_init_cb(1341) > Call init_cb for firmwareUpdate capability
value is NULL
value is NULL
I (8861) [IoT]: st_cap_send_attr(613) > publish event, topic : /v1/deviceEvents/6543a7c7-2d1c-4e86-8675-ba35fea2a1e9, payload :
{"deviceEvents":[{"component":"main","capability":"firmwareUpdate","attribute":"currentVersion","value":"switch_example_001","providerData":{"sequenceNumber":2,"timestamp":"1708535020063"}}]}
Sequence number return : 2
value is NULL
value is NULL
value is NULL
status: 8, stat: 2
I (8891) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x208
I (9111) [IoT]: _iot_parse_noti_data(650) > payload : {"target":"6543a7c7-2d1c-4e86-8675-ba35fea2a1e9","values":{},"event":"device.preferences"}
I (9111) [IoT]: _iot_parse_noti_data(749) > No references
I (9121) [IoT]: iot_noti_sub_cb(829) > Ignore notification
E (32701) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
I (32701) esp-tls-mbedtls: Certificate verified.
E (32701) esp-tls: Failed to open new connection
E (32701) transport_base: Failed to open a new connection
E (32711) HTTP_CLIENT: Connection failed, sock < 0
HTTP_EVENT_ERROR
Failed to open HTTP connection: 28674

LUSAN54, 2024-02-23:
I have mounted another servir with a dedicated linux ubuntu PC and the error is the same.
E (32701) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
I (32701) esp-tls-mbedtls: Certificate verified.
E (32701) esp-tls: Failed to open new connection
E (32701) transport_base: Failed to open a new connection
E (32711) HTTP_CLIENT: Connection failed, sock < 0
HTTP_EVENT_ERROR
Failed to open HTTP connection: 28674
I need help, i can´t continue my develop.

junyoun-kim, 2024-02-26:
Hello @LUSAN54 Did you make sure copy root.pem under ota_dmo/main folder? You should overwrite root.pem of your server. After that, please clean and rebuild demo app.

LUSAN54, 2024-02-26:
i´m sure i have copied de root.pem under ota_demo/main folder.
I don´t understand "You should overwrite root.mep of your server".

**I have repetead all the procces.**
In the linux server:
1.- Generate  root.key, root.csr and root.crt
2.-Generate server.crs, server.key and server sever.crt.
3.-Generate public_key.pem and root.pem
4.-I have copied public_key.pem and root.pem to my ota_demo /main
5.- I have build and flash my ota_demo.bin 
6.-I have modified the file "device_info.json" changing  "firmwareVersion"
7.-i have build the new version and copy de file .bin to the server to sing it with "root.key".
8.-i have copied server.crt and server .key to the server ota_demo directory and version_info.json and signed_ota_demo.bin to the /ota_demo/dir
9.- i have run the scritpt python3 ../https_server.py and the sudo iptables -I INPUT 1 -p tcp --dport 4443 -j ACCEPT command
10.- i test the connection with the server witch "curl --cacert ./root.crt "https://192.168.1.3:4443"" and the result is ok, i can see the server ota_demo/dir from another linux machine.
11.-My device works fine with the smarththings app and I can turn my device on and off from the app.
12.-But it can´t connect with the ota_demo server. error messages are still:
jBCTEcifQ==.xSjC1eaPmB08YXSMfFKekWALykzztGe++Zq6fU8XH7UnETcjOrJ0Udr4ZsKx+Q/j4sHHJB2XzhG2uqciRU1bBg==
I (7422) [IoT]: _iot_net_tls_connect(416) > Loading the CA root certificate 1317@0x3ffcd328
I (8132) [IoT]: iot_es_connect(1106) > MQTT connect success sucess/try : 1/1
I (8232) [IoT]: _check_connection_response(79) > Connection response payload {"target":"bc3eb29e-3396-4823-8bba-b15e43d4731f","currentTime":1708941752,"event":"connect.success"}
I (8242) [IoT]: _do_iot_main_command(586) > curr_main_cmd:0, curr_main_state:5
I (8252) [IoT]: _do_state_updating(435) > current state 5, new state 6
I (8252) [IoT]: _get_device_preference(423) > Get device preference
I (8262) [IoT]: iot_cap_call_init_cb(1341) > Call init_cb for switch capability
I (8272) [IoT]: st_cap_send_attr(613) > publish event, topic : /v1/deviceEvents/bc3eb29e-3396-4823-8bba-b15e43d4731f, payload :
{"deviceEvents":[{"component":"main","capability":"switch","attribute":"switch","value":"on","providerData":{"sequenceNumber":1,"timestamp":"1708941752496"}}]}
Sequence number return : 1
I (8302) [IoT]: iot_cap_call_init_cb(1341) > Call init_cb for firmwareUpdate capability
value is NULL
value is NULL
I (8312) [IoT]: st_cap_send_attr(613) > publish event, topic : /v1/deviceEvents/bc3eb29e-3396-4823-8bba-b15e43d4731f, payload :
{"deviceEvents":[{"component":"main","capability":"firmwareUpdate","attribute":"currentVersion","value":"switch_example_001","providerData":{"sequenceNumber":2,"timestamp":"1708941752536"}}]}
Sequence number return : 2
value is NULL
value is NULL
value is NULL
status: 8, stat: 2
I (8342) [IoT]: _do_state_updating(573) > Call usr status_cb with 0x208
I (8562) [IoT]: _iot_parse_noti_data(650) > payload : {"target":"bc3eb29e-3396-4823-8bba-b15e43d4731f","values":{},"event":"device.preferences"}
I (8562) [IoT]: _iot_parse_noti_data(749) > No references
I (8572) [IoT]: iot_noti_sub_cb(829) > Ignore notification
E (32712) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
I (32712) esp-tls-mbedtls: Certificate verified.
E (32712) esp-tls: Failed to open new connection
E (32722) transport_base: Failed to open a new connection
E (32732) HTTP_CLIENT: Connection failed, sock < 0
HTTP_EVENT_ERROR
Failed to open HTTP connection: 28674

The only strange thing I can see is that when I run $ python3 ../https_server.py
my linux terminal terminal shows this message:

lusan@lusan-linux:~/ota_demo/dir$ python3 ../https_server.py
/home/lusan/ota_demo/dir/../https_server.py:10: DeprecationWarning: ssl.wrap_socket() is deprecated, use SSLContext.wrap_socket()
  httpd.socket = ssl.wrap_socket (httpd.socket,
192.168.67.190 - - [26/Feb/2024 10:22:48] "GET / HTTP/1.1" 200 -

junyoun-kim, 2024-02-26:
Hmm.. I think esp32 mbedtls is returning -0x7780 which is MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE. Maybe it's related with your esp32 mbedtls config and server tls config. Could you turn on mbedtls debug log and test it again?

You can turn debug log on with below sequence.
> ./build.py esp32 ota_demo menuconfig

In menu select page,
Component config -> mbedTLS -> Enable mbedTLS debugging

junyoun-kim, 2024-02-27:
Hmm.. We are facing another problem by enabling mbedtls debug. Due to lots of log, device can't connect to Cloud server(SmartThings server) because of timeout. To look into your local server connecting issue, could you delete below line and test it again??

under void ota_polling_task(void *arg) funcing In main.c
delete below code,
line 132 ~ 134
        if (g_iot_status != IOT_STATUS_CONNECTING || g_iot_stat_lv != IOT_STAT_LV_DONE) {
            continue;
        }

LUSAN54, 2024-02-27:
Goor morning, see attached the new logs after deleting lines 132 to 134.
[log_270224.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/14416649/log_270224.txt)

junyoun-kim, 2024-02-27:
@LUSAN54 one thing to check. Did you change below IP info with your local server IP?

In main/ota_util.c file
`#define CONFIG_OTA_SERVER_URL "https://192.168.1.3:4443/"`

LUSAN54, 2024-02-27:
yes, sure:

#define CONFIG_OTA_SERVER_URL "https://192.168.67.249:4443/"

junyoun-kim, 2024-02-27:
From the log, looks like your local server refusing connection during ssl handshake. It can be they can't agree on connection setting problem.

```
 I (32778) mbedtls: ssl_tls.c:3939 => handshake
 I (32788) mbedtls: ssl_msg.c:2124 => flush output
 I (32798) mbedtls: ssl_msg.c:2133 <= flush output
 I (32818) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_HELLO_REQUEST
 I (32828) mbedtls: ssl_msg.c:2124 => flush output
 I (32848) mbedtls: ssl_msg.c:2133 <= flush output
 I (32858) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_CLIENT_HELLO
 I (32888) mbedtls: ssl_client.c:906 => write client hello
 I (32908) mbedtls: ssl_msg.c:2554 => write handshake message
 I (32918) mbedtls: ssl_msg.c:2714 => write record
 I (32928) mbedtls: ssl_msg.c:2851 <= write record
 I (32948) mbedtls: ssl_msg.c:2675 <= write handshake message
 I (32968) mbedtls: ssl_client.c:994 <= write client hello
 I (32968) mbedtls: ssl_msg.c:2124 => flush output
 I (32978) mbedtls: ssl_msg.c:2138 message length: 133, out_left: 133
 I (32988) mbedtls: ssl_msg.c:2145 ssl->f_send() returned 133 (-0xffffff7b)
 I (32998) mbedtls: ssl_msg.c:2172 <= flush output
 I (33028) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_HELLO
 I (33038) mbedtls: ssl_tls12_client.c:1195 => parse server hello
 I (33058) mbedtls: ssl_msg.c:3887 => read record
 I (33078) mbedtls: ssl_msg.c:1926 => fetch input
 I (33078) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5
 I (33108) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5
 I (33108) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
 I (33118) mbedtls: ssl_msg.c:2111 <= fetch input
 I (33128) mbedtls: ssl_msg.c:1926 => fetch input
 I (33148) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 7
 I (33178) mbedtls: ssl_msg.c:2086 in_left: 5, nb_want: 7
 I (33198) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 2 (-0xfffffffe)
 I (33218) mbedtls: ssl_msg.c:2111 <= fetch input
 I (33228) mbedtls: ssl_msg.c:4792 got an alert message, type: [2:40]
 W (33238) mbedtls: ssl_msg.c:4799 is a fatal alert message (msg 40)
 W (33268) mbedtls: ssl_msg.c:3942 mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
 W (33288) mbedtls: ssl_tls12_client.c:1199 mbedtls_ssl_read_record() returned -30592 (-0x7780)
 I (33308) mbedtls: ssl_tls.c:3950 <= handshake
 E (33328) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
 I (33338) esp-tls-mbedtls: Certificate verified.
```

Could you check why you local httpd server refusing connection if there is server log?

Also could you check local server support tls 1.2? Maybe you could test it with below command
`curl --tlsv1.2 --cacert ./root.crt "https://192.168.67.35:4443/"`

junyoun-kim, 2024-02-27:
Okay.. Can you use tcpdump command at your local server? If then, can you share tcpdump pcap output file between your server and esp32?

You can refer below sites about how to use tcpdump.
https://www.techtarget.com/searchnetworking/tutorial/How-to-capture-and-analyze-traffic-with-tcpdump
https://www.baeldung.com/linux/tcpdump-capture-ssl-handshake

LUSAN54, 2024-02-27:
See attached the log between server and esp32.
Can you see anything?
[log tcpdump.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/14424189/log.tcpdump.txt)
lusan@lusan-linux:~$ sudo tcpdump src host 192.168.67.249 and tcp port 4443 or src host 192.168.67.22
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eno2, link-type EN10MB (Ethernet), snapshot length 262144 bytes
19:18:36.458466 ARP, Request who-has desktop-o15tt10.dinuy.local tell fichadormecanica.dinuy.local, length 46
19:18:36.458770 IP fichadormecanica.dinuy.local.52657 > desktop-o15tt10.dinuy.local.4443: Flags [S], seq 1825322306, win 5840, options [mss 1436], length 0
19:18:36.458819 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [S.], seq 3278029797, ack 1825322307, win 64240, options [mss 1460], length 0
19:18:37.475353 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [S.], seq 3278029797, ack 1825322307, win 64240, options [mss 1460], length 0
19:18:37.635291 IP fichadormecanica.dinuy.local.52657 > desktop-o15tt10.dinuy.local.4443: Flags [.], ack 1, win 5840, length 0
19:18:39.332635 IP fichadormecanica.dinuy.local.52657 > desktop-o15tt10.dinuy.local.4443: Flags [P.], seq 1:134, ack 1, win 5840, length 133
19:18:39.332691 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [.], ack 134, win 64107, length 0
19:18:39.332925 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [P.], seq 1:8, ack 134, win 64107, length 7
19:18:39.333085 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [F.], seq 8, ack 134, win 64107, length 0
19:18:39.359614 IP fichadormecanica.dinuy.local.52657 > desktop-o15tt10.dinuy.local.4443: Flags [.], ack 9, win 5832, length 0
19:18:39.392938 IP fichadormecanica.dinuy.local.52657 > desktop-o15tt10.dinuy.local.4443: Flags [F.], seq 134, ack 9, win 5832, length 0
19:18:39.392967 IP desktop-o15tt10.dinuy.local.4443 > fichadormecanica.dinuy.local.52657: Flags [.], ack 135, win 64107, length 0

junyoun-kim, 2024-02-28:
Could you run below command and test again with esp32? and share pcap output file?

`tcpdump "tcp port 4443" -w client-hello.pcap`

junyoun-kim, 2024-02-28:
From pcap you shared, below is  cipher suites esp32(client) sent.

[image]

Maybe your local server don't support any cipher suite among those.

You can select esp32 supporting cipher suites from menuconfig -> Component config -> mbedTLS.
(Looks like currently esp32 is supporting cipher suites only RSA key exchange.
You can try it with another key exchange method like DHE_RSA or ELLIPTIC_CURVE)

To find out which cipher suites your local server supports, I think you can get lists from below python script at your local server.
```
import ssl

ctx = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
ctx.get_ciphers()
```

Or, If you share success case(your local server <-> linux) pcap file, let me check which cipher they used.

LUSAN54, 2024-02-28:
Or, If you share success case(your local server <-> another computer client) pcap file, let me check which cipher they used.
See attached the success case between linux machine and server:
[client-hell2.zip](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/14432566/client-hell2.zip)

junyoun-kim, 2024-02-28:
Looks like cipher suite they are using is `TLS_AES_256_GCM_SHA384` which is TLS 1.3 feature.

[image]

I'm not sure esp32 support the TLS 1.3 cipher suite, but you can explore at esp32 mbedTLS component menuconfig.

Also there must be some TLS 1.2 cipher suites local server supporting. Like I said above, you can try it with other key exchange method like DHE_RSA or ELLIPTIC_CURVE not RSA.

LUSAN54, 2024-02-28:
Bingo!
Changing to DHE_RSA or ELLIPTIC_CURVE "NOT" RSA. Now it is ok.

LUSAN54, 2024-02-28:
I have a new problem.
After updating correctly the switch_sample_001 to version switch_sample_002, i try to update version switch_sample_002 to switch_sample_003 version. For that, i change de files device_info.json changing de version to "switch_sample_003". I build the file, sign the file and put de new file in the server, after that i change the file version_info.json in the server like this:
{
  "versioninfo": {
  "latest": "switch_example_003",
  "upgrade": ["switch_example_002"],
  "polling": "1"
  }
}
Wen i reset my device , the device inicialize and i see that the device connect with the server ( i see the log) but the smarThign app not sow me that it has a new version and i can´t actualize the device.

What I can be doing wrong?

LUSAN54, 2024-02-28:
Now is OK.
i have migrated the switch_example in ESP32 to the ESP32C3 and it works to.
Thank you very much for your attention and support.

LUSAN54, 2024-03-25:
> Enviado el: miércoles, 28 de febrero de 2024 16:03
>
> As I know, SmartThings app blocks new firmware version update available for 1 hour since the last firmware update.
> So you need to wait 1 hour to test next firmware update. To skip this, you may need to flush out SmartThings app data and cache in App Setting and relaunch SmartThings App.

Today I have a new problem.
After several months of testing my device, today the mobile application gives me this error:

Can you help me?

junyoun-kim, 2024-03-26:
Could you update error code again? It doesn't appear.

junyoun-kim, 2024-11-27:
Hello, @LUSAN54   Is there any more questions? We will close this issue if no more updates.

closed 2024-12-02 by junyoun-kim