SmartThingsCommunity/st-device-sdk-c-ref #149: Encrypting stnv bin partition

asierdanjou, 2024-04-23:
Hi, 

I'm trying to modify esp32c3 "switch_example" to encrypt s/n and paired keys in stnv partition.

I've got working the example right in both modes, taking keys from device.JSON file and taking them from stnv partition (with flashing the keys BIN file in STNV partition). Now I want to encrypt this BIN file. To do that I follow next steps:

1. I generate encrypted BIN file next way:
“…/nvs_partition_gen.py encrypt input.csv out_enc.bin --version 1 --keygen”

2. Then I flash encrypted bin to STNV partition doing next:
“python $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32c3 write_flash 0x9000 out_enc.bin”

3.  Finally, I flash bin file with the keys to decript STNV partition. I flash them in nvs_keys partition:
“…parttool.py --partition-table-offset 0x8000 write_partition --partition-name=“nvs_key” --input keys-04-18_09-59.bin”

I put my ESP32c3 running but it fails. It seems that it doesn't get to connect with smartthings server.

Are those steps correct? Do I have to change something in "switch_example" code to initializate STNV as secure mode?

Any help would be greatly appreciated.

Best regards,

Asier.

junyoun-kim, 2024-04-24:
@asierdanjou Hello, Could you share device log? We want to check which step fails to connect smartthings server.

asierdanjou, 2024-04-24:
@junyoun-kim thank you for your reply.
I attach the file with de log we get in monitor log.
[asierdanjou_log.txt](https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/files/15090572/asierdanjou_log.txt)

asierdanjou, 2024-04-25:
Hi,
I've got the solution to my doubs.
It seems that it is not possible to enable "NVS encryption" if we do not previously activate "flash encryption on boot" option in the menuconfig.
I enable both and get "switch_example" to work with encrypted smartthings credentials.
Thank you for your time,
Asier.

closed 2024-05-02 by asierdanjou