SmartThingsCommunity/st-device-sdk-c-ref #151: How to reflash nvs-key partition when flash encryption is enabled

asierdanjou, 2024-05-07:
Hi,
I'm trying to flash nvs-key partition again when "encrypted flash on boot" option is enabled.
I add two options to parttool.py:
--esptool-write-args encrypt
--partition-table-file partitions.2MB.csv

```
asier@DESKTOP-F9FS6J3:~/st-device-sdk-c-ref/apps/esp32c3/switch_example/output_bulk/20240417_190809/TEST01/stnv/keys$ python /home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py --esptool-write-args encrypt --partition-table-offset 0xA000 --partition-table-file partitions.2MB.csv write_partition --partition-name="nvs_key" --input keys-04-18_09-59.bin

Running /home/asier/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/asier/esp/esp-idf/components/esptool_py/esptool/esptool.py erase_region 90112 4096...
esptool.py v4.7.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 48:27:e2:ad:91:cc
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Active security features detected, erasing flash is disabled as a safety measure. Use --force to override, please use with caution, otherwise it may brick your device!
An exception: ** Command '['/home/asier/.espressif/python_env/idf5.2_py3.10_env/bin/python', '/home/asier/esp/esp-idf/components/esptool_py/esptool/esptool.py', 'erase_region', '90112', '4096']' returned non-zero exit status 2. ** occurred in _call_esptool.
Traceback (most recent call last):
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 365, in <module>
    main()
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 358, in main
    op(**common_args)
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 187, in _write_partition
    target.write_partition(partition_id, input)
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 173, in write_partition
    self.erase_partition(partition_id)
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 166, in erase_partition
    self._call_esptool(['erase_region', str(partition.offset),  str(partition.size)] + self.esptool_erase_args)
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 141, in _call_esptool
    raise e
  File "/home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/partition_table/parttool.py", line 138, in _call_esptool
    subprocess.check_call(esptool_args, stdout=out, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/asier/.espressif/python_env/idf5.2_py3.10_env/bin/python', '/home/asier/esp/esp-idf/components/esptool_py/esptool/esptool.py', 'erase_region', '90112', '4096']' returned non-zero exit status 2.
```
How should I do it?
Thanks,
Asier

junyoun-kim, 2024-05-08:
How about using --force option as instruction said.

> A fatal error occurred: Active security features detected, erasing flash is disabled as a safety measure. Use --force to override, please use with caution, otherwise it may brick your device!

asierdanjou, 2024-05-08:
Hi,
I can't find any documentation for the --force argument for parttool.py, nor for esptool.py.
Is there any example of its use ?

Anyway I tried the following:

```
asier@DESKTOP-F9FS6J3:~/st-device-sdk-c-ref/apps/esp32c3/switch_example_plaintext_ota_sinJSON_enc/output_bulk/20240417_190809/TEST01/stnv/keys$ python /home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/esptool_py/esptool/esptool.py erase_region 0x16000 0x1000
esptool.py v3.1-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 4)
Features: Wi-Fi
Crystal is 40MHz
MAC: 48:27:e2:ad:91:cc
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.1 seconds.
Hard resetting via RTS pin...

asier@DESKTOP-F9FS6J3:~/st-device-sdk-c-ref/apps/esp32c3/switch_example_plaintext_ota_sinJSON_enc/output_bulk/20240417_190809/TEST01/stnv/keys$ python /home/asier/st-device-sdk-c-ref/bsp/esp32c3/components/esptool_py/esptool/esptool.py write_flash 0x16000 keys-04-18_09-59.bin --encrypt
esptool.py v3.1-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 4)
Features: Wi-Fi
Crystal is 40MHz
MAC: 48:27:e2:ad:91:cc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00016000 to 0x00016fff...

WARNING: - compress and encrypt options are mutually exclusive
Will flash keys-04-18_09-59.bin uncompressed
Wrote 16384 bytes at 0x00016000 in 1.5 seconds (86.2 kbit/s)...

Leaving...
Hard resetting via RTS pin...
```

But after doing that it doesn't work either.

Asier.

asierdanjou, 2024-06-07:
Hi,
Any help with that?
Thank you,
Asier.

jonggab-park, 2024-06-11:
Sorry, it's too late.
We start to preparing the guide for nvs encryption.
If we complete the guide, we will update.

Hwang-JeongHo, 2024-07-23:
We have added a guide on how to encrypt nvs in esp32.
I hope this guide will help you with your issue.

https://github.com/SmartThingsCommunity/st-device-sdk-c-ref/blob/develop/doc/nvs_encryption_esp32.md

junyoun-kim, 2024-11-27:
Hello, @asierdanjou  Is there any more questions? We will close this issue if no more updates.

asierdanjou, 2024-11-27:
Hi junyoun-kim,
No more questions.
Thank you very much for your time.

closed 2024-11-27 by asierdanjou